ophub/amlogic-s9xxx-openwrt #2: Error after installing luci-proto-qmi

latiefkhan, 2021-01-01:
Here the error message.

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/network/iface_add'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/network/proto_qmi.lua:53: attempt to call method 'register_error_code' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

ophub, 2021-01-01:
Do you use a 4G network card for wan Internet access? Since your USB network card is not included when the firmware is compiled, it cannot be recognized by default. You can add support for your USB network card after fork. Enter the USB network card in the router_config/.config file (SSH connection to Actions is recommended)

make menuconfig：
Base system  ---> udev 
Kernel modules  ---> USB Support  ---> kmod-usb-acm kmod-usb-net kmod-usb-net-qmi-wwan kmod-usb-ohci kmod-usb-serial kmod-usb-serial-option L
uCI  ---> Protocols  ---> luci-proto-3g luci-proto-qmi 
Network  ---> uqmi 
Utilities  ---> usb-modeswitch usbutils

make kernel_menuconfig：
device drivers->usb support->support for host-side usb SELECT USB Modem (CDC ACM) support

USB Serial Converter support：
USB driver for GSM and CDMA modems

ophub, 2021-01-01:
http://h-wrt.com/en/mini-how-to/qmi_ow

latiefkhan, 2021-01-01:
Iam using Quallcomm usb modem 4g LTE device,
Okay i will try to make menuconfig

latiefkhan, 2021-01-01:
it's still doesn't work sir. :(
Iam so borred. How to resolve this :(

latiefkhan, 2021-01-01:
And i doesn't use ubuntu desktop.
Iam using windows 10,
 so please make the firmware pacakage included what i need sir.

ophub, 2021-01-01:
1.fork to your repository.

2.On the Action page. Select Build OpenWrt in the list on the left, and Click the Run workflow button on the right. Set SSH connection to Actions: **true** to use tmate to connect to the GitHub Actions virtual server environment. 

You can directly perform the make menuconfig operation to generate the compilation configuration, or any customized operation. After triggering the workflow, wait for the SSH connection to Actions step to be executed on the Actions page, and then the following three lines of messages will be displayed: 1. To connect to this session copy-n-paste the following into a terminal or browser, 2.ssh Y26QenMRd@nyc1.tmate.io, 3.https://tmate.io/t/Y26QenMRd. Then copy the SSH connection command and paste it into the terminal for execution, or copy the link to open it in the browser and use the web terminal. enter the command: cd openwrt && make menuconfig for personalized configuration (The web terminal may encounter a black screen, just press Ctrl+C). After completion, press the shortcut key Ctrl+D or execute the exit command to exit, and the subsequent compilation work will proceed automatically.

ophub, 2021-01-01:
Fork the latest repository source code
I added a personalized selection of network equipment to the configuration file /router_config/.config. You can check the specific model of your equipment, select the suitable one, and compile it.

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/.config
in 207 line # WWAN
in 217 line # Network Devices

Add the plugin source you imported in /router_config/diy-part2.sh
Add your plug-in name luci-proto-qmi=y in luci-app of /router_config/.config

Just compile

ophub, 2021-01-01:
How To use LTE modem in QMI mode for WAN connection

https://openwrt.org/docs/guide-user/network/wan/wwan/ltedongle

latiefkhan, 2021-01-01:
> 
> 
> How To use LTE modem in QMI mode for WAN connection
> 
> https://openwrt.org/docs/guide-user/network/wan/wwan/ltedongle

that's tutorial is make same error too sir

ophub, 2021-01-01:
This tutorial fully introduces how to use it. It should be possible to choose the correct driver for the device.

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/.config
in 207 line # WWAN
in 217 line # Network Devices

latiefkhan, 2021-01-01:
> 
> 
> This tutorial fully introduces how to use it. It should be possible to choose the correct driver for the device.
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/.config
> in 207 line # WWAN
> in 217 line # Network Devices

yes i hope this one will work, iam still compiling the file

latiefkhan, 2021-01-01:
have you done released new version of openwrt hk1 box.?
but, why i unnable to download the file..?

ophub, 2021-01-01:
Compiling, download in Release after 3 hours

latiefkhan, 2021-01-01:
> Compiling, download in Release after 3 hours

including my request problem.?

ophub, 2021-01-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/455868305

Compiling, download after 3 hours

ophub, 2021-01-01:
> > Compiling, download in Release after 3 hours
> 
> including my request problem.?

Add LTE support according to the tutorial on the openwrt official website, you can try to install  luci-proto-qmi. Not sure if it matches the model of your device. You can modify it in the .config file

latiefkhan, 2021-01-01:
same problem sir when installing luci-proto-qmi
[image]

ophub, 2021-01-01:
> same problem sir when installing luci-proto-qmi
> [image]

After three hours, test whether the firmware with LTE support can be used normally. Which is your device? Find the same paragraph at https://www.ebay.com/ and send it.

latiefkhan, 2021-01-01:
https://www.ebay.com/itm/T77W676-917817-001-lt4220-X12-LTE-4G-WWAN-M-2-450Mbps-Modem-1030-hp-EliteX-G4P6/133173804804?hash=item1f01c6ff04:g:jBoAAOSwMXddgB3c

latiefkhan, 2021-01-01:
this log of modem on my s905x running well

[image]

latiefkhan, 2021-01-01:
but i have no idea, why in my HK1 box is always getting error :(

ophub, 2021-01-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/openwrt_s9xxx_2021.01.01.1907

Compilation is complete, you can download this version to test

latiefkhan, 2021-01-01:
has been tested, and same result sir. after installing luci-proto-qmi, and accessing- network-interface "same error
[image]

ophub, 2021-01-01:
Is your luci-proto-qmi the latest compatible openwrt R20.12.12? Does this plugin need other dependent packages?

ophub, 2021-01-01:
https://openwrt.org/packages/pkgdata/luci-proto-qmi
release: OpenWrt-19.07.0

You can select the specified version to compile on line 64 of https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt.yml

latiefkhan, 2021-01-01:
> 
> 
> Is your luci-proto-qmi the latest compatible openwrt R20.12.12? Does this plugin need other dependent packages?

iam installing luci-proto-qmi
[image]
 using this method

latiefkhan, 2021-01-01:
now, iam done installing luci-proto-qmi, but the new problem is, any protocol i have installed mannualy, isn't showing up.
[image]
[image]

ophub, 2021-01-01:
What agreement? Newly installed protocols generally need to be restarted to take effect. Send the name, let me see if it needs to be selected at compile time.

latiefkhan, 2021-01-01:
> What agreement? Newly installed protocols generally need to be restarted to take effect. Send the name, let me see if it needs to be selected at compile time.

it must be showing 'Qmi Cellular protocol., Because the luci-proto-qmi has been successfully installed.

ophub, 2021-01-02:
> > What agreement? Newly installed protocols generally need to be restarted to take effect. Send the name, let me see if it needs to be selected at compile time.
> 
> it must be showing ''Qmi Cellular'' protocol., Because the luci-proto-qmi has been successfully installed.
> Restarted Hk1 box is not work sir. Still not showing up.

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/.config
Added relevant Protocols to support luci-proto in line 290, try again after compilation is complete.

The compilation process:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/456972031

latiefkhan, 2021-01-02:
this update is verry perfect my LTE CAT12 modem Running well sir, verry amazing. thank you so much for helping me sir.

latiefkhan, 2021-01-02:
issues closed!!

closed 2021-01-02 by latiefkhan