ophub/amlogic-s9xxx-openwrt #5: Install on sdcard

riyuejz, 2021-01-04:
how to run with a bootable sdcard, on another amlogic device.. device: hg-680p

latiefkhan, 2021-01-04:
Edit dtb file using windows xplorer, adding meson is already exist.

ophub, 2021-01-05:
Supported range：

https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/armbian/dtb-amlogic

Log in to the default IP: 192.168.1.1 → Login in to openwrt → system menu → TTYD terminal → input command:

s9xxx-install.sh

The s9xxx-v*-openwrt_*.img is the general OpenWrt firmware of all Amlogic-S9xxx series boxes . You can write this OpenWrt firmware to the USB hard disk to start. When writing into EMMC through s9xxx-install.sh, select the name of the box you own in the menu.

For more OpenWrt firmware .dtb files are in the dtb-amlogic directory. You can use the s9xxx-v*-openwrt_*.img firmware to install via USB hard disk. When writing into EMMC through s9xxx-install.sh, select 0: Enter the dtb file name of your box, and use the S9xxx-Boxes you own.

closed 2021-01-06 by ophub