ophub/amlogic-s9xxx-openwrt #6: Installing on internal emmc is not stable internet connection

latiefkhan, 2021-01-04:
When iam testing my internet speed using my hk1box, and wan connection using 4G LTE modem, and the firmware installed on emmc internal, the speed internet is only 50Mbps,
When i boot from sdcard and running the firmware on sdcard, testing my speed internet, i got 140Mbps, so great installing on sd card, but bad idea to write in internal emmc.

ophub, 2021-01-05:
Run the following command in the shell, no need to restart, take effect immediately
Enter advanced mode:

echo "performance" > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor

ophub, 2021-01-05:
The real reason for the sometimes poor performance of the ARM64 box running speedtest may have been found, which is affected by the energy-saving mode.
In Arm64's op firmware, the default power mode is "ondemand", which usually runs at a lower frequency, and only increases the CPU frequency when necessary (and overclocking has a certain delay), and speedtest is a short-time high-burst mode For speed measurement, the speed measurement is often completed before the CPU frequency is raised, which creates the false impression of low speed measurement results, which is not a problem with other speed measurement tools.
Therefore, if you want to run scores or speed tests, you need to make the cpu enter the high-performance mode "performance" in advance.

ophub, 2021-01-05:
You can change the default value to performance in CPU free
login openwrt > services > cpu frep > CPU Freq Settings > CPU Scaling Governor > select performance

latiefkhan, 2021-01-05:
> You can change the default value to performance in CPU free
> login openwrt > services > cpu frep > CPU Freq Settings > CPU Scaling Governor > select performance
> 
> The reason for the sometimes poor performance of the ARM64 box running speedtest is affected by the energy-saving mode. In Arm64's op firmware, the default power mode is "ondemand", which usually runs at a lower frequency, and only increases the CPU frequency when necessary (and overclocking has a certain delay), and speedtest is a short-time high-burst mode For speed measurement, the speed measurement is often completed before the CPU frequency is raised, which creates the false impression of low speed measurement results, which is not a problem with other speed measurement tools.
> Therefore, if you want to run scores or speed tests, you need to make the cpu enter the high-performance mode "performance" in advance.

Oke i will try running high performance,

closed 2021-01-05 by latiefkhan