ophub/amlogic-s9xxx-openwrt #8: Wifi driver Amlogic-S9xxx series

riyuejz, 2021-01-17:
Please add wifi driver "rtl8189fs" on Amlogic-S9xxx series

ophub, 2021-01-19:
There is no integrated rtl8189fs driver in openwrt of lede. You can import and compile from a third party.

1. Find a suitable driver, such as:
`https://github.com/ap17/rtl8189fs`

2. Add to the compiler, such as:
Enter the [https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/diy-part2.sh](diy-part2.sh) file and add the following code:
`git clone https://github.com/ap17/rtl8189fs.git package/rtl8189fs`

3. Enable SSH remote connection, manually select the rtl8189fs driver, enter
Select `Build OpenWrt` on the `Actions` page. Click the `Run workflow` button. SSH connection to Actions: `true`

4. Observe the action process. In the SSH connection to Actions step, press / enter `rtl8189fs` to find the path of the driver, and select it according to this path.

5. Exit ssh to compile

riyuejz, 2021-01-19:
Can you make it for me .. I really don't know what to do

ophub, 2021-01-19:
I tried to compile it, The rtl8189fs driver is not compatible with the current openwrt kernel. I'm sorry I can't help you.

closed 2021-01-20 by ophub