ophub/amlogic-s9xxx-openwrt #10: s905x release images?

Falcordog, 2021-02-08:
I cannot find any openwrt_s905x_v*.img releases although they are mentioned in the documentation.
Am I missing something?

ophub, 2021-02-09:
Select `Build OpenWrt` on the `Actions` page. Click the ` Run workflow ` button.

Wait for the compilation to be completed, the `  openwrt-armvirt-64-default-rootfs.tar.gz ` file compiled this time will be automatically uploaded in the ` Releases `

If there is an file in a Releases in your repository, you can use this file to directly package the required firmware.

eg: https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/openwrt_s9xxx_2021.02.09.0324

ophub, 2021-02-09:
`s905x` and `s905d`, `n1` have different firmware names, but the firmware is `the same`, you can use s905d or N1 firmware. The correspondence between other models and the same firmware of the box is as [supported-models-and-boxes](https://github.com/ophub/amlogic-s9xxx-openwrt#supported-models-and-boxes). 

You can also add the `s905x` parameter in line 161 of the [.github/workflows/build-openwrt.yml](https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt.yml), and the firmware named `openwrt_s905x_v*.img` will be obtained after compilation.

Falcordog, 2021-02-09:
Understood.
Thanks for clarifying!

closed 2021-02-09 by Falcordog