ophub/amlogic-s9xxx-openwrt #9: Is S905M-B supported?

xiaofami, 2021-02-08:
My BestTv R3300-M box ships with S905M-B soc with 1G ram and 4G emmc . Currently I'm running Armbian_5.44_S9xxx_Ubuntu_bionic_3.14.29_server_20180729.img without issue ( with gxbb_p200_1G_100M_RealtekWiFi.dtb) . However the later Armbian releases from Balbe150 seems have dropped support since there's no gxbb_p200_1G_100M_RealtekWiFi.dtb and neither could meson-gxbb-p200.dtb or meson-gxbb-p201.dtb boot my R3300-M box . Any chance for this box to run openwrt with 4.X or 5.X kernel ? I have been starring at my dts file for days trying to figure out something but still pointless :(

I extracted dtb file from R3300-M android 4.4 system and converted to txt format,hope to help .

[R3300_M.txt](https://github.com/ophub/amlogic-s9xxx-openwrt/files/5942217/R3300_M.txt)

ophub, 2021-02-08:
`meson-gxbb-p201.dtb` should correspond to your model, if it cannot be used, there is no other dtb file for now.

closed 2021-02-10 by ophub

xiaofami, 2022-03-07:
> > My BestTv R3300-M box ships with S905M-B soc with 1G ram and 4G emmc . Currently I'm running Armbian_5.44_S9xxx_Ubuntu_bionic_3.14.29_server_20180729.img without issue ( with gxbb_p200_1G_100M_RealtekWiFi.dtb) . However the later Armbian releases from Balbe150 seems have dropped support since there's no gxbb_p200_1G_100M_RealtekWiFi.dtb and neither could meson-gxbb-p200.dtb or meson-gxbb-p201.dtb boot my R3300-M box . Any chance for this box to run openwrt with 4.X or 5.X kernel ? I have been starring at my dts file for days trying to figure out something but still pointless :(
> > I extracted dtb file from R3300-M android 4.4 system and converted to txt format,hope to help .
> > [R3300_M.txt](https://github.com/ophub/amlogic-s9xxx-openwrt/files/5942217/R3300_M.txt)
> 
> https://tccmu.com/2021/03/24/R3300-M/ 老哥这是你的网站吧？最后咋样了？

网站是我的 XD 后来测试恩山上flippy为MXQ制作的镜像使用没问题，不过百兆网卡是瓶颈，由于镜像里没编译rtl8189es驱动WIFI不可用，此外还不错。另外这个盒子运行Manjaro ARM还挺满意，网站上写过几篇不赘述了

ophub, 2022-03-07:
8189es在内核里有，你编译内核的时候选择即可。

https://github.com/unifreq/linux-5.15.y/blob/5ce83fdb69a6288486490e7db932f50716c36cdb/drivers/net/wireless/realtek/rtl8189fs/Makefile#L255