ophub/amlogic-s9xxx-openwrt #11: luci-openvpn

riyuejz, 2021-02-21:
[image]

after installing luci-openvpn openvpn-openssl I tried to add the ovpn config profile instead like this ... s905x hg680p

ophub, 2021-02-22:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/588047358

I just added openvpn to the package, and I will test it after the compilation.

riyuejz, 2021-02-22:
ok I will wait for the results

riyuejz, 2021-02-22:
don't forget to add a new ovpn client config whether it works or not .. thanks for all the help 🙏

ophub, 2021-02-22:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/.config

I added the client and server in lines 333 and 334, and tested it after the compilation was completed.

ophub, 2021-02-22:
Compilation is complete, I tested Services > OpenVPN, and  the VPN > OpenVPN-Server,All the pages are displayed normally.

ophub, 2021-02-22:
router_config/.config
lines 333 and 334

CONFIG_PACKAGE_luci-app-openvpn=y
CONFIG_PACKAGE_luci-app-openvpn-server=y

riyuejz, 2021-02-22:
Have you tried adding .ovpn configuration to openvpn client manually? the page is an error if you add the OVPN configuration manually ...

ophub, 2021-02-22:
[image]

I don’t know how to use openvpn, but I added one based on his default, but no errors were reported.

riyuejz, 2021-02-22:
[image]
thanks .. this version works well 🙏

closed 2021-02-22 by riyuejz