ophub/amlogic-s9xxx-openwrt #13: ZTE B860H(S905x) Wifi not detected

aarzaary, 2021-02-24:
The wifi driver is rtl8189fs

ophub, 2021-02-24:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/595812726

I tried to add this driver, you can wait for the completion of the compilation to test, write emmc after the first start, wait 5 minutes for wifi, refresh the network menu and then take a look. Please give feedback about your test results.

aarzaary, 2021-02-24:
Okay, thanks
I'll try after the build is complete

aarzaary, 2021-02-24:
[image]
Still not detected :(
Kernel not supported yet?

ophub, 2021-02-24:
```

cd /root
wget https://loksado.com/files/hg680p/5.3.0/rtl8189fs.ko
mv -f rtl8189fs.ko /lib/modules/*/kernel/drivers/net/wireless/
root@loksado:~# sudo depmod -a
root@loksado:~# sudo modprobe rtl8189fs

```

aarzaary, 2021-02-24:
-ash: depmod: not found

aarzaary, 2021-02-25:
After I checked it turned out that the driver only supports kernel 5.3.0
Are there any releases with kernel 5.30?

ophub, 2021-02-25:
The openwrt installed in the STB uses the Armbian kernel, so theoretically the drivers that can be used by Armbian can also be used in openwrt. I found some Armbians suitable for this STB on the Internet. The WIFI drivers in the same path can be copied to openwrt by reference Use the same path as, because there is no such device at hand, you can try: 

```
#Download Armbian Linux Amlogic S905x untuk HG680P dan B860H:
https://loksado.com/2020/02/download-armbian-linux-untuk-hg680p-dan-b860h-amlogic-s905-p212/

#Mengaktifkan Wifi Internal HG680-P di Armbian
https://loksado.com/2019/08/mengaktifkan-wifi-internal-hg680-p-di-armbian/

#Armbian Linux dengan Driver Wifi Internal Untuk HG680-P
https://loksado.com/2020/06/armbian-20-05-6-dengan-driver-wifi-internal-untuk-hg680-p/

#Driver Wifi Internal HG680-P untuk Armbian 19.11.3
https://loksado.com/2020/02/driver-wifi-internal-hg680-p-untuk-armbian-19-11-3/

#Mengenal Perbedaan dan Ciri HG680-P Hitam dan Putih
https://loksado.com/2020/04/mengenal-perbedaan-dan-ciri-hg680-p-hitam-dan-putih/

#The driver file I found is placed in the wireless directory, you can test it 
[wireless](https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/build_kernel/patches/root/wireless)

```

aarzaary, 2021-02-25:
I have try put *.ko file from arm it's doesn't work
I think because:
1. Kernel version is different
2. Architecture diffirent b860h (aarch64) & hg680p (arm)

I'm try install *.ko file and look at dmesg it's say error about version of kernel & architecture

I'm found driver for aarch64 but only for kernel 5.3.0
And openwet with kernel 5.3.0 not available at
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/kernel

One more my question
Is possible use armbian linux kernel to boot openwrt?

If it's not please close this issue

ophub, 2021-02-25:
STB originally did not have openwrt, it is not a router, so the official website of openwrt will not support STB.

The openwrt you have seen is to extract the Armbian kernel, and then replace/merge the relevant files of openwrt to make an openwrt suitable for use in the STB.

I provide a script to extract the kernel from other people’s openwrt, or extract the kernel from armbian, you can see what is extracted from it, and make your own kernel:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/build_kernel

Where `make_use_img.sh` is to extract the packaged kernel from the existing `armbian` or `openwrt` firmware.

cp -rf boo/{dtb,config*,initrd.img*,System.map*,uInitrd,zImage} ${kernel_tmp}  #for kernel.tar.xz
cp -rf root/lib/modules ${modules_tmp}  #for modules.tar.xz

Extract these two parts, you can generate a different kernel packaging material, put it in the `kernel/your_kernel` directory, and then use `sudo ./make -d -b s905x -k your_kernel` to package

closed 2021-02-25 by ophub