ophub/amlogic-s9xxx-openwrt #17: Req...

riyuejz, 2021-02-26:
next update can you add the rclone package from https://github.com/ElonH/Rclone-OpenWrt 🙏

ophub, 2021-02-26:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/.config
I added this plug-in on `line 404`.
`CONFIG_PACKAGE_luci-app-rclone=y`

the new firmware is being compiled, please pay attention. 
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/602000028

ramasedang, 2021-02-26:
@ophub so helpfull, tanks bro

ramasedang, 2021-02-26:
So, if I want to use package in the list I must recompile it? Or just opkg install...

ophub, 2021-02-26:
luci-app-rclone has been compiled to the latest firmware, you can download and install.

Other packages that are not available can also be installed online like this: 

System > Package > Update, then select the package and install it.
Most of the packages that are not precompiled can be installed in this way, and some need to install dependent packages first.

The software package to be used is recommended to be selected at compile time.

ramasedang, 2021-02-26:
"System > Package > Update, then select the package and install it.
Most of the packages that are not precompiled can be installed in this way, and some need to install dependent packages first."

ophub, 2021-02-26:
[image]

ophub, 2021-02-26:
[image]

ramasedang, 2021-02-26:
@ophub 
i try instal luci-app-softethervpn...... 

 "You can add them by yourself. For example, modify # CONFIG_PACKAGE_luci-app-acme is not set to CONFIG_PACKAGE_luci-app-acme=y"

where i can edit the config? in firmware or before compiled

[image]

[image]

ophub, 2021-02-26:
[image]

Click the `fork` button to copy my repo to your personal center, and then you can compile it in your own repo. For related functions, see the instructions on the official github.com websit.

https://docs.github.com/en/github/getting-started-with-github/fork-a-repo

ramasedang, 2021-02-26:
Thanks for your help 🙏

ophub, 2021-02-26:
Some software can be installed directly in the mirror server. For example, after clicking the update list, search for `luci-app-nlbwmon`, and this software package will have.

The mirror server can also retrieve the latest version of the installed software package, depending on the mirror server you set up. There are many openwrt mirror sites in various countries. You can find it on Google.com

ophub, 2021-02-26:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router_config

Here is a description of personalized customization.

In the readme.md of various places, I have introduced each function as detailed as possible. You can check the readme.md in each module. If there is something unclear, you can open an `issue` at any time.

closed 2021-02-26 by ophub