ophub/amlogic-s9xxx-openwrt #20: [ASK] New Branch for Default OpenWrt + Luci + Docker only

dianariyanto, 2021-02-26:
Hello sir,

You make an Awesome firmware running on my s905x. But, have a lot of packages that I don't need. Can you make a new branch for default Openwrt with Luci only? or maybe Openwrt + Luci + Docker? 

I think that can make a good fast and simple firmware for my s905x.

Thanks, for your awesome work.

ophub, 2021-02-26:
Hello there,
Some friends are submitting the need to add packages by default, and some want to delete some, so it is difficult to make it suitable for everyone. In the warehouse, the default configuration of the software package can be customized. The specific configuration is in router_config, and the choice of software source is in the .config file. If you want to import other software sources, use diy-part1.sh and diy-part1. sh to introduce.

You click the fork button in the upper right corner of the warehouse to copy a copy of the warehouse source code, and you can customize it in your personal center.

If you don't understand the specific operation process, you can mention it. I will write a more detailed and personalized firmware customization operation instruction.

dianariyanto, 2021-02-26:
> Hello there,
> Some friends are submitting the need to add packages by default, and some want to delete some, so it is difficult to make it suitable for everyone. In the warehouse, the default [...]

You are right but will be more awesome if it has a stock Openwrt build and Ophub customized version. I didn't think all people will need these preinstalled apps on it.

I'm not too familiar with github-action yet, and it's still very new on github. Maybe I need a lot to learn to build openwrt firmware for s905x from scratch. Can I build these images using openwrt imagebuilder?

https://openwrt.org/docs/guide-user/additional-software/imagebuilder

dianariyanto, 2021-02-26:
Hi @ophub 

I've already forked this repo, then can I just remove all of this config to build a clean openwrt+luci?
https://github.com/dianariyanto/amlogic-s9xxx-openwrt/blob/main/router_config/.config

ophub, 2021-02-27:
https://github.com/dianariyanto/amlogic-s9xxx-openwrt/blob/stock/router_config/.config

Lines 418-535 are bilateral
```
- Required (no # in front and = y in the back) CONFIG_PACKAGE_luci-app-cjdns = y

- Don’t need it (there is # before it is not set): #CONFIG_PACKAGE_luci-app-cjdns is not set
```

dianariyanto, 2021-02-27:
@ophub 

That's great! 

Already make some changes and now it building using Github-actions. Realy cool, let see if it work. Thanks for your awesome help sir.

closed 2021-02-27 by ophub