ophub/amlogic-s9xxx-openwrt #21: Cannot satisfy the following dependencies for xxxxx

dianariyanto, 2021-02-28:
Unable to install app from package repository

```bash
root@OpenWrt:/# opkg install luci-app-mwan3
Installing luci-app-mwan3 (git-20.356.64372-1259bb1-1) to root...
Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.8/packages/aarch64_cortex-a53/luci/luci-app-mwan3_git-20.356.64372-1259bb1-1_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-mwan3:
 *      kernel (= 5.4.100-1-37e10f03f2389d0610b948c6e12e3215)
 * opkg_install_cmd: Cannot install package luci-app-mwan3.
```

ophub, 2021-03-01:
This package has been included, you can use the following opkg command to view.

```
opkg update
opkg list-installed | grep mwan
```

View other installed packages: `opkg list-installed`

dianariyanto, 2021-03-01:
Sorry, I think it's about kernel mismatch.

closed 2021-03-01 by dianariyanto