ophub/amlogic-s9xxx-openwrt #23: Extracting kernel from another openwrt with multiple kernel

sajati, 2021-03-04:
Hi, i tried to extract kernel from this file https://sfile.mobi/8BKWAvNPMcd , but probably there are 2 version of kernel in single openwrt firmware image because when i open directory of rootfs partition `/lib/modules` there are 2 folders `4.14.195` and `5.7.2-arm-64` and results in error when using `make_use_img.sh` ( only extract the `4.14.195` version, same as when using [tuanqing/mknop](https://github.com/tuanqing/mknop)'s `sudo ./gen_openwrt -e`
is there any workaround to make `make_use_img.sh` extract the `5.7.2-arm-64` version instead of `4.14.195` version ?
i do this because on that openwrt image, rtl8189fs wireless driver is working, and i see some issues here noticed wireless problem on hg680p, if it done it may solve that problems
thanks.

ophub, 2021-03-04:
The current firmware is developed for `Flippy`, I will first test and extract it from this firmware, and then I will reply to you.

ophub, 2021-03-04:
I added rtl8188fu and rtl8189fs wifi to the three versions compiled today. You can test whether these three versions can drive normally. Pay attention to this compilation, download and test after completion.

https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/620011538

ramasedang, 2021-03-04:
How do I enable wifi, I've tried 3 different kernels but no changes...

I think you can make version 18 firmware, bcs kernel 3 can use wifi on hg680p and b860h

[image]

ophub, 2021-03-04:
Under the network menu, is there a wireless network?

ramasedang, 2021-03-04:
> Under the network menu, is there a wireless network?

I cannot find that

ophub, 2021-03-04:
My openwrt-armvirt-64-default-rootfs.tar.gz is compiled using https://github.com/coolsnowwolf/lede, I extracted the firmware you shared, the luci-page opens and the interface displays an error. It may be necessary to compile openwrt-armvirt-64-default-rootfs.tar.gz from the source of the openwrt.org official website. I will try tomorrow which source of the firmware you shared. You can use the https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/amlogic-s9xxx/amlogic-kernel/build_kernel/make_use_img.sh to extract the kernel.

ophub, 2021-03-04:
Wait a minute, I will post the kernel file extracted from the openwrt firmware you shared

ramasedang, 2021-03-04:
> Wait a minute, I will post the kernel file extracted from the openwrt firmware you shared

> Wait a minute, I will post the kernel file extracted from the openwrt firmware you shared

Okay

sajati, 2021-03-04:
> I added rtl8188fu and rtl8189fs wifi to the three versions compiled today. You can test whether these three versions can drive normally. Pay attention to this compilation, download and test after completion.
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/620011538

I got these logs :
```
Thu Mar  4 14:03:54 2021 kern.err kernel: [   20.716328] 8189fs: version magic '5.7.2-arm-64 SMP preempt mod_unload aarch64' should be '5.9.16-flippy-51+ SMP mod_unload aarch64'
Thu Mar  4 14:03:54 2021 kern.err kernel: [   20.735096] rtl8188fu: version magic '5.3.0-aml-g12 SMP preempt mod_unload aarch64' should be '5.9.16-flippy-51+ SMP mod_unload aarch64'
Thu Mar  4 14:03:54 2021 user.err kernel: [   20.748382] kmodloader: 2 modules could not be probed
Thu Mar  4 14:03:54 2021 user.err kernel: [   20.752828] kmodloader: - 8189fs - 0
Thu Mar  4 14:03:54 2021 user.err kernel: [   20.758438] kmodloader: - rtl8188fu - 0
```
Still no `Wireless` under `Network`

sajati, 2021-03-04:
> I think you can make version 18 firmware, bcs kernel 3 can use wifi on hg680p and b860h

njir..  wegare udah bikin 19.04 kernel 5 lu malah mau 18 kernel 3

ophub, 2021-03-04:
I extracted the kernel from https://sfile.mobi/8BKWAvNPMcd:  5.7.2w
It is recommended that you compile an `openwrt-armvirt-64-default-rootfs.tar.gz`  from `https://github.com/openwrt/openwrt/tree/openwrt-19.07` or other source code libraries of the `19.07.4`
`sudo ./make -d -b s905x -k 5.7.2w`

ramasedang, 2021-03-04:
> > I think you can make version 18 firmware, bcs kernel 3 can use wifi on hg680p and b860h
> 
> njir.. wegare udah bikin 19.04 kernel 5 lu malah mau 18 kernel 3

> > I think you can make version 18 firmware, bcs kernel 3 can use wifi on hg680p and b860h
> 
> njir.. wegare udah bikin 19.04 kernel 5 lu malah mau 18 kernel 3

Wowkwowo bukan gw pak, gw user 19, gw req buat mewakili org org yg suka bilang "kok WiFi gamuncul ya" Klo gw sih Pake acces point, enak nya fw ini ada SSR++ nya coba bisa di extract jadiin IPK enak kali yak :'

sajati, 2021-03-04:
> I extracted the kernel from https://sfile.mobi/8BKWAvNPMcd.
> [Kernel path: 5.7.2w](https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/kernel/5.7.2w)
> It is recommended that you compile an `openwrt-armvirt-64-default-rootfs.tar.gz` from https://github.com/Lienol/openwrt or other source code libraries of the `19.07.4`
> `sudo ./make -d -b s905x -k 5.7.2w`

[image]
I'm pulling it now, my internet speed is below average and using WSL under windows 10
i'm trying to compile [ImmortalWrt](https://github.com/immortalwrt/immortalwrt)
i already do some, just the problem on wifi.
i will post the result here later

ophub, 2021-03-04:
https://github.com/immortalwrt/immortalwrt/tree/openwrt-19.07

use `19.07`

It should be because this kernel is extracted from `19.07`, I don't know if other branches show errors in LUCI.

sajati, 2021-03-04:
[image]

Now `Wireless` submenu is there under `Network`, but dont know why the SSID is not casting , not discoverable
[image]

ophub, 2021-03-04:
I found a 19.07 in this place: https://github.com/Lienol/openwrt
but I am not familiar with this source code library, I try to compile with his default configuration to see if it can be successfully compiled to `openwrt-armvirt-64-default-rootfs.tar.gz`

See the compilation result: https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/621190788

ophub, 2021-03-04:
Ah, this is an improvement. At least I have seen the wireless network. How to configure it further and change the frequency to the default? Check the related setting items.

sajati, 2021-03-04:
> Ah, this is an improvement. At least I have seen the wireless network. How to configure it further and change the frequency to the default? Check the related setting items.

i have this log 20+ times, don't know what to do, actually i'm new in linux :
```
Thu Mar  4 21:37:47 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 21:37:47 2021 daemon.err modprobe: - bpfilter
```
i will try to compile 19.07.4 version, the same as wegare did and see if the wireless is working...
will post the result here later

ophub, 2021-03-04:
https://github.com/Lienol/openwrt
19.07
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/621190788

#---------
https://github.com/immortalwrt/immortalwrt
openwrt-19.07
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/621237880

After the two source code libraries of the 19.07 branch are compiled, test whether they can be used.

ramasedang, 2021-03-04:
> i have this log 20+ times, don't know what to do, actually i'm new in linux :
>
> ```
> Thu Mar  4 21:37:47 2021 daemon.err modprobe: 1 module could not be probed
> Thu Mar  4 21:37:47 2021 daemon.err [...]

Nyimak bang :(

sajati, 2021-03-04:
> Nyimak bang :(

Pusing wkwkwk, undangin suhu wegare lah...

sajati, 2021-03-04:
so i quick build a rootfs image with [openwrt 19.07.4 image builder ](https://downloads.openwrt.org/releases/19.07.4/targets/armvirt/64/openwrt-imagebuilder-19.07.4-armvirt-64.Linux-x86_64.tar.xz) with the same packages as wegare's firmware except 4 unnoficial packages like simple-obfs and shadowsocks which not exist in openwrt repository, still the same, wireless submenu is shown but undiscoverable, cannot scan as well ( trying to scan my mobile phone hotspot ).

seeing this in kernel log :
```
[   22.399293] rtl8188fu: loading out-of-tree module taints kernel.
[   22.457053] rtl8188fu: Unknown symbol cfg80211_inform_bss_frame_data (err -2)
[   22.462993] rtl8188fu: Unknown symbol cfg80211_scan_done (err -2)
[   22.468991] rtl8188fu: Unknown symbol cfg80211_remain_on_channel_expired (err -2)
[   22.476423] rtl8188fu: Unknown symbol cfg80211_new_sta (err -2)
[   22.482246] rtl8188fu: Unknown symbol cfg80211_disconnected (err -2)
[   22.488549] rtl8188fu: Unknown symbol wiphy_new_nm (err -2)
[   22.494049] rtl8188fu: Unknown symbol cfg80211_ready_on_channel (err -2)
[   22.500731] rtl8188fu: Unknown symbol wiphy_register (err -2)
[   22.506401] rtl8188fu: Unknown symbol cfg80211_put_bss (err -2)
[   22.512253] rtl8188fu: Unknown symbol cfg80211_roamed (err -2)
[   22.518121] rtl8188fu: Unknown symbol cfg80211_ibss_joined (err -2)
[   22.524290] rtl8188fu: Unknown symbol cfg80211_michael_mic_failure (err -2)
[   22.531199] rtl8188fu: Unknown symbol wiphy_apply_custom_regulatory (err -2)
[   22.538171] rtl8188fu: Unknown symbol ieee80211_get_channel (err -2)
[   22.544421] rtl8188fu: Unknown symbol cfg80211_del_sta_sinfo (err -2)
[   22.550817] rtl8188fu: Unknown symbol wiphy_unregister (err -2)
[   22.556703] rtl8188fu: Unknown symbol cfg80211_get_bss (err -2)
[   22.562581] rtl8188fu: Unknown symbol cfg80211_mgmt_tx_status (err -2)
[   22.569059] rtl8188fu: Unknown symbol cfg80211_rx_mgmt (err -2)
[   22.574932] rtl8188fu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[   22.581963] rtl8188fu: Unknown symbol cfg80211_connect_done (err -2)
[   22.588235] rtl8188fu: Unknown symbol cfg80211_unlink_bss (err -2)
[   22.594359] rtl8188fu: Unknown symbol wiphy_free (err -2)
```

and this on system log :
```
Thu Mar  4 22:54:35 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:35 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:35 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:35 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:35 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:35 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:36 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:36 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:36 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:36 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:36 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:36 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:36 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:36 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:37 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:37 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:37 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:37 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:37 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:37 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:38 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:38 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:38 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:38 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:38 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:38 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:39 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:39 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:39 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:39 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:39 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:39 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:40 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:40 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:40 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:40 2021 daemon.err modprobe: - bpfilter
Thu Mar  4 22:54:40 2021 daemon.err modprobe: 1 module could not be probed
Thu Mar  4 22:54:40 2021 daemon.err modprobe: - bpfilter
```

i also copied the config files `/etc/config/*` of wegare's firmware to the new firmware but didn't make sense

ramasedang, 2021-03-05:
> > Nyimak bang :(
> 
> Pusing wkwkwk, undangin suhu wegare lah...

bagi immortalwrt nya ngab

ophub, 2021-03-06:
Sorry, copying the driver directly may not work, the driver may need to be compiled when the kernel is generated.

closed 2021-03-06 by ophub

nawirwii, 2021-04-05:
nyimakkk ahhh..