ophub/amlogic-s9xxx-openwrt #24: X96 max + turn black screen

johnmalay, 2021-03-09:
After flashing openwrt image and reboot the box. The box then become black screen. There is no network activity detect after plug in lan cable. Before flash I'm successfully boot from USB. I set network config within my network ip. I can browse luci web. But after installing into emmc and reboot. The box won't give any respond. Just black screen. I plug lan cable in to pc, no ip given to lan adapter. It just show unidentified network. Give the manual ip and ssh using putty also can't.

johnmalay, 2021-03-10:
Stb = x96 max +
Firmware = https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_2021.03.09.1944/openwrt_s905x3_v5.4.101_2021.03.09.1933.img.gz
Connecting monitor just show black screen. Trying to reset with reset button inside av port. Nothing happen.

ophub, 2021-03-10:
- Connect the monitor, there is no code prompt at all? Are there any code hints, please take a screenshot? 
- I just tested and installed the same firmware, it can be used normally after writing.
-  you can see the screenshot of my installation and the screenshot after restart

[image]
[image]

johnmalay, 2021-03-10:
[image]
Stuck just like that. No code showing. I just run openwrt-install no other command such as TEST-UBOOT. Sorry if I miss the step and brick my device

ophub, 2021-03-10:
If you installed the v5.10.19.TF version, this yesterday's version forgot to make the judgment of restricting writing to EMMC, and it cannot be started after installation (it does not affect the reinsertion of the U disk, booting from the U disk), you can download it again If there is no .TF, restart the installation on the USB flash drive.

You try to insert the 5.4.101 version of the USB hard drive and reinstall.

ophub, 2021-03-10:
Which country are you from, I will help you find an initialization tutorial.
Can you read Chinese? I have seen many tutorials written by Chinese people on the Internet. It would be convenient if you can read Chinese, or you can use Google Translate to help you understand.

johnmalay, 2021-03-10:
I live in Malaysia and can read Indonesian. I'm using non tf image . Live boot from USB. I try to reinstall but the problem is my monitor does not show anything. Just blank screen. No error message. I'm stuck. Yesterday I buy usb male to male to flash stock firmware, open up the burning tools, but the usb not detected. Try usb 2.0 & 3.0 also stuck. I don't know how to fix the problem. Sorry to disturbing you. 🙏

ophub, 2021-03-10:
https://zhuanlan.zhihu.com/p/293943099

This help contains the software and firmware to restore and initialize the TV system. You can read it with the help of google translation. I have used this solution when testing the firmware, it is perfectly restored to the original TV system, and then you can reinstall openwrt.

johnmalay, 2021-03-10:
Tq my friend. I follow the guide and successfully restore stock firmware. Short the circuit do the trick. Now to confirm which firmware should I use to flash again without any problem. I run install-openwrt inside the live usb boot. I do not run the command inside ttyd at luci. Is it the cause of the problem? I see successful flash before restarting and the problem occurs.

johnmalay, 2021-03-10:
[image]
Which one should I select? Yesterday build I select number 1 and after that stb brick

johnmalay, 2021-03-10:
[image]
Yesterday build I run live usb. Check the luci page it's working. After flashing into memory then it brick

ophub, 2021-03-10:
按Installation command: openwrt-install

Choose 1, install it

ophub, 2021-03-10:
I added two possible problems and recovery methods in the installation document:

https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/install-program

closed 2021-03-10 by ophub