ophub/amlogic-s9xxx-openwrt #26: Change luci-app-openvpn

ikhsanh, 2021-03-12:
Hi, Can you use luci-app-openvpn with this version? https://downloads.openwrt.org/releases/19.07.6/packages/aarch64_cortex-a53/luci/luci-app-openvpn_git-21.070.25463-b4025cf-1_all.ipk
Cause  I'm confused about how to use the OpenVPN feature in this OpenWrt firmware I can't import an OpenVPN file. I tried to install it myself, but maybe there is a firewall that needs to be set up but I don't know how, so that OpenVPN can't connect. sorry for my bad English

ophub, 2021-03-12:
This package is already integrated.

[image]

ikhsanh, 2021-03-12:
Ok. I will try.

closed 2021-03-12 by ikhsanh