ophub/amlogic-s9xxx-openwrt #27: Pirania

adisubagja, 2021-03-13:
Can add this to openwrt sir?

https://github.com/libremesh/pirania

https://github.com/libremesh/lime-packages/tree/master/packages/pirania-app
https://github.com/libremesh/lime-packages/tree/master/packages/pirania

ophub, 2021-03-14:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/diy-part2.sh

fork repository.
On lines 27 and 28, delete the # at the beginning of the line, like this:

```
git clone https://github.com/libremesh/lime-packages.git package/lime-packages
sed -i "/DEFAULT_PACKAGES/ s/$/ pirania-app pirania ip6tables-mod-nat ipset/" target/linux/armvirt/Makefile
```
Build your personalized plug-in. More usage methods can be found in this document:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/Documentation.md

adisubagja, 2021-03-14:
Like this ? https://github.com/adisubagja/amlogic-s9xxx-openwrt/blob/main/router_config/diy-part2.sh#L27

iam using b860h this no problem for target armvirt?

ophub, 2021-03-14:
yes

adisubagja, 2021-03-14:
Thanks i'll try

adisubagja, 2021-03-14:
Im done build with this, but now how to check pirania is installed bro?

ophub, 2021-03-14:
E.g:

cat /etc/config/pirania

Check whether the file of this software exists, and find the relevant location according to the official instructions.

https://github.com/libremesh/lime-packages/tree/master/packages/pirania

```
view files:
    /etc/config/pirania is the UCI config
    /etc/pirania/db.csv (default path) contains the database of vouchers

    /etc/init.d/pirania-uhttpd starts a uhttpd on port 59080 that replies any request with a redirect towards a preset URL

    /usr/lib/lua/voucher/ contains lua libraries used by /usr/bin/voucher
    /usr/bin/voucher is a CLI to manage the db (has functions add_voucher, add_many_vouchers, auth_voucher, get_valid_macs, list_vouchers, remove_voucher and url)
    /usr/bin/captive-portal sets up iptables rules to capture traffic

    /usr/libexec/rpcd/pirania ubus pirania API (this is used by the web frontend)
    /usr/share/rpcd/acl.d/pirania.json ACL for the ubus pirania API
```

ophub, 2021-03-15:
In addition to the entire repository integration, you can also choose the required package integration:
```
svn co https://github.com/libremesh/lime-packages/trunk/packages/{shared-state-pirania,pirania-app,pirania} package/lime-packages/packages
sed -i "/DEFAULT_PACKAGES/ s/$/ pirania-app pirania ip6tables-mod-nat ipset shared-state-pirania uhttpd-mod-lua/" target/linux/armvirt/Makefile
```

My two lines of code are how openwrt adds third-party plug-in sources and compiles them.

I use `SSH connection to Actions` to check that these two packages have been selected, so the addition method is effective.

[image]

I did not find their integrated compilation instructions on this plugin website, you can ask them how to compile/install/use them.

adisubagja, 2021-03-15:
Yes im installing my fw build but not showing this file :(
I don't know what else to do
If I find any more ideas I will tell again

Thank you for the help

closed 2021-03-15 by adisubagja