ophub/amlogic-s9xxx-openwrt #35: How to add this

adisubagja, 2021-03-19:
How to include [xderm-mini GUI](https://github.com/ryanfauzi1/xderm-mini_GUI)  by default

ophub, 2021-03-19:
The other party does not have a standard compilation interface and can only execute the installation script manually. You can consider adding these scripts at the end of openwrt-install

adisubagja, 2021-03-20:
Thanks
I will trying install my build tommorow

Im using b860h
my wifi is offline i think this error from radio0 not enable / missing radio can ur help this?

ophub, 2021-03-20:
The wifi of this STB cannot be driven temporarily. We tried to add it several times but failed. The driver cannot be used in openwrt.

adisubagja, 2021-03-20:
when I use kernel 3.x wifi can run perfectly maybe there is something to do sir?

ophub, 2021-03-20:
https://github.com/ophub/build-armbian/blob/main/kernel.sh
I am trying to compile armbian and make the kernel here, but I just started to try it. If you are familiar with kernel compilation, you can make and use it according to the method of kernel extraction.

adisubagja, 2021-03-20:
Ok lemme read for learn sir
Thanks before 🙂

closed 2021-03-20 by ophub