ophub/amlogic-s9xxx-openwrt #36: Switch and upstream

madIlama, 2021-03-20:
Hello,nice work you have in this repo
I just finished flashing ur image and want to ask:

1. There is no switch support for s905?(mine is s905x), either swdev or dsa ?

2. You base on snapshot, can we get stable base like 19.04 or 19.07 to use opkg without manually(or automatically) pre-build ?

ophub, 2021-03-20:
What do you mean by `swdev or dsa`? Please send the full name, or the address of the other party’s repo, and I will see if the other party provides an integrated compilation method for openwrt.

Here is how to add third-party software packages, including how to use opkg:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/README.md

madIlama, 2021-03-20:
What i mean is this

https://openwrt.org/docs/guide-user/network/vlan/switch_configuration#is_there_a_vlan-capable_hardware_switch_integrated_in_your_device

https://openwrt.org/docs/guide-user/network/vlan/switch

ophub, 2021-03-20:
Do you ask again, how to compile other openwrt source code libraries? Such as using the official openwrt to compile? For example, use other branches such as 19.04 or 19.07?

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt.yml
```
#Line 24: Source code library address
REPO_URL: https://github.com/coolsnowwolf/lede
#Line 25: Branch name
REPO_BRANCH: master
```

You can modify it to other, such as official
```
REPO_URL: https://github.com/openwrt/openwrt
REPO_BRANCH: openwrt-19.07
```

madIlama, 2021-03-20:
I see,thats for branch.

But how about switch/vlan ? It seems related to kernel

ophub, 2021-03-20:
`Switch/vlan` is the basic function of the router, `lean` uses the basis of `openwrt`, and has made extensions such as software packages. You can use `/etc/config/network` to view

The basic router functions are exactly the same:
https://github.com/coolsnowwolf/lede/blob/master/feeds.conf.default

madIlama, 2021-03-20:
> `Switch/vlan` is the basic function of the router, `lean` uses the basis of `openwrt`, and has made extensions such as software packages. You can use `vi /etc/config/network` to view
> 
> The basic router functions are exactly the same:
> https://github.com/coolsnowwolf/lede/blob/master/feeds.conf.default
> 
> You can use the official openwrt tutorial to configure: https://openwrt.org/docs/start

Ok,so from what i read this repo again based on lean openwrt and flippy kernel

While lean repo based on snapshot.

I have been noticed,there is lienol repo in your credit,can i use lienol in workflow?

ophub, 2021-03-20:
absolutely okay

closed 2021-03-20 by ophub