ophub/amlogic-s9xxx-openwrt #38: Workflow fails with openwrt official repo

madIlama, 2021-03-25:
I try to build using official openwrt with 21.02 branch
It stopped at this point

```
Run [ -d openwrt-armvirt ] || mkdir -p openwrt-armvirt
cp: cannot stat 'openwrt/bin/targets/*/*/*.tar.gz': No such file or directory
Welcome to use the OpenWrt packaging tool!

 Filesystem     Type      Size  Used Avail Use% Mounted on
udev           devtmpfs  3.4G     0  3.4G   0% /dev
tmpfs          tmpfs     696M  676K  695M   1% /run
/dev/sdb1      ext4       84G   28G   56G  33% /
tmpfs          tmpfs     3.4G  8.0K  3.4G   1% /dev/shm
tmpfs          tmpfs     5.0M     0  5.0M   0% /run/lock
tmpfs          tmpfs     3.4G     0  3.4G   0% /sys/fs/cgroup
/dev/sdb15     vfat      105M  6.1M   99M   6% /boot/efi
/dev/sda1      ext4       14G  4.1G  9.0G  32% /mnt 

 [  Error  ] No the [ openwrt-armvirt-64-default-rootfs.tar.gz ] file in [ /home/runner/work/amlogic-s9xxx-openwrt/amlogic-s9xxx-openwrt/openwrt-armvirt ] directory!
Error: Process completed with exit code 1.
```

ophub, 2021-03-25:
Using the openwrt with 21.02 branch  official repositories, many `luci-apps` may not be compiled. The current configuration is `coolsnowwolf/lede`, I need to test it.

ophub, 2021-03-27:
openwrt-21.02 is born for you:

REPO_URL: https://github.com/openwrt/openwrt
REPO_BRANCH: openwrt-21.02
ROUTER_CONFIG: https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config/openwrt-21.02
WORKFLOWS: https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt-21.02.yml

closed 2021-03-27 by ophub