ophub/amlogic-s9xxx-openwrt #39: Error opkg update

adisubagja, 2021-03-26:
root@OpenWrt:~# opkg update
Collected errors:
 * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
[image]

So Im Using OpenVPN Connection and getting this how to fix?

ophub, 2021-03-26:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router_config/.config#L417

`luci-app-openvpn`  has been integrated in the firmware, but I don’t know how to use the `openvpn`  plug-in. You can search for how to configure it. I tested `opkg` locally and it can be used normally.

adisubagja, 2021-03-26:
Im using custom openvpn config using terminal
Like file ovpn & login.txt
at line auth-user-pass put login.txt

like example :
auth-user-pass login.txt

then put username & password in login.txt file

this my method to run openvpn
just type `openvpn file.ovpn &`

i dont know using self config in luci-app-openvpn :( if u have contact like telegram,whatsapp,etc i'll chat to solve this

ophub, 2021-03-26:
Someone asked how to use this `luci-app-openvpn` plugin before. But I will not use this plugin.

We use BBM in Indonesia, but my mom doesn't let me play mobile phones.

adisubagja, 2021-03-26:
where u from bro? and you have pc bro?
i think telegram desktop available on pc

ophub, 2021-03-26:
Telegram cannot be used in Indonesia, we use BBM.

closed 2021-03-26 by ophub