ophub/amlogic-s9xxx-openwrt #42: Samba setup

riyuejz, 2021-03-28:
can you help me to set it up .. i'm using hdd ntfs partition .. so I can read and write how to do it..
[image]
[image]
[image]

ophub, 2021-03-28:
[image]

riyuejz, 2021-03-28:
[image]
Invalid SMB status: STATUS_ACCESS_DENIED

ophub, 2021-03-29:
https://forum.openwrt.org/t/how-i-got-samba4-working-for-me/68633
https://openwrt.org/docs/guide-user/services/nas/samba_configuration

closed 2021-03-29 by ophub