ophub/amlogic-s9xxx-openwrt #47: Add a third-party software package

lynxnexy, 2021-04-07:
can you please help me to add this pkg into the build config.. 🙏
pkg: https://github.com/lutfailham96/libernet

ophub, 2021-04-09:
I checked their instructions and only provided a manual installation method, not a compilation method. Missing `Makefile`

ramasedang, 2021-04-09:
> I checked their instructions, only provided a manual installation method, not a compilation method. Missing `Makefile`
can u add openclash? i try install openclash, if i use ssr+ firmware openclash its normal but if i use passwall firmware the page of openclash open slowly, always loading

ophub, 2021-04-09:
I did not find it in the source code repository of lede . Please provide the repository address of this plugin

ophub, 2021-04-09:
Can these plugins be opened at the same time? You can only open one at the same time, and close the others, otherwise there will be conflicts.

closed 2021-04-10 by ophub