ophub/amlogic-s9xxx-openwrt #49: Problem with installing kmod driver

elbarfallah, 2021-04-10:
Hello, I want to install kmod-ath9k-htc, but opkg said that it isn't compatible with the architectures configured. Previously in LEDE version, I can install that driver by using argument --force-depends. But from this official version, it doesn't work anymore.
Here is the log:

root@OpenWrt:~# opkg install kmod-ath9k-htc
Unknown package 'kmod-ath9k-htc'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ath9k-common found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ath9k-htc found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ath9k-htc.

root@OpenWrt:~# opkg install kmod-ath9k-htc --force-depends
Unknown package 'kmod-ath9k-htc'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ath9k-common found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ath9k-htc found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ath9k-htc.

ophub, 2021-04-10:
`CONFIG_PACKAGE_kmod-ath9k-htc=y`

I set the default integration of this driver, you can try again after the compilation is complete

elbarfallah, 2021-04-10:
[image]
Thanks, Now, I can use my usb wifi for my stb.

closed 2021-04-10 by elbarfallah