ophub/amlogic-s9xxx-openwrt #50: make script error

sajati, 2021-04-10:
```
 [  s905x  -  5.10.26.TF  ]  extract armbian files.
grep: etc/init.d/boot: No such file or directory
sed: can't read etc/init.d/boot: No such file or directory
grep: etc/init.d/boot: No such file or directory
sed: can't read etc/init.d/boot: No such file or directory
sed: can't read etc/inittab: No such file or directory
sed: can't read etc/inittab: No such file or directory
./make: line 294: cd: usr/lib/lua/luci/view/admin_status: No such file or directory
 [  s905x  -  5.10.26.TF  ]  make openwrt image.
 [  s905x  -  5.10.26.TF  ]  copy files to image.
 [  s905x  -  5.10.26.TF  ]  generate success.
```

ophub, 2021-04-10:
Synchronize the latest source code, your files is incomplete.
What system are you using?

sajati, 2021-04-10:
I'm using ubuntu under windows 10 WSL, when i `git pull` it said `already up to date`

also i'm using this command when cloning `git clone --depth 1 --branch openwrt_s9xxx_official_2021.04.10.0600 https://github.com/ophub/amlogic-s9xxx-openwrt.git /home/${USER}/ready-tools/amlogic-s9xxx-openwrt` is it possible to cause that issue ?

ophub, 2021-04-11:
`git clone --depth 1 https://github.com/ophub/amlogic-s9xxx-openwrt.git`

Use this clone repository command

closed 2021-04-12 by ophub