ophub/amlogic-s9xxx-openwrt #53: wont boot to emmc

Chicadaa, 2021-04-13:
Hello, I already install openwrt to emmc (B860H V1) and when I tried to reboot why it won't boot to the openwrt?
sorry for my bad english

ophub, 2021-04-13:
Dial the power source, and then try to plug in the power source. What is the version number of the kernel you installed.

Chicadaa, 2021-04-13:
How do i check it?

Chicadaa, 2021-04-13:
Do I need to change boot file?

ophub, 2021-04-13:
meson-gxl-s905x-p212.dtb

Chicadaa, 2021-04-13:
Done, What to do next?

ophub, 2021-04-13:
Then it's all right. Unplug the U disk, unplug the power supply, plug in the power supply again, wait one minute for the system to start, and you can use it

Chicadaa, 2021-04-13:
> Then it's all right. Unplug the U disk, unplug the power supply, plug in the power supply again, wait one minute for the system to start, and you can use it

I did, but why the system won't start

ophub, 2021-04-13:
Start and end your installation. Send screenshots of these two parts. Let me take a look at your installation.

Chicadaa, 2021-04-13:
[image]
[image]
[logs.txt](https://github.com/ophub/amlogic-s9xxx-openwrt/files/6303947/logs.txt)

ophub, 2021-04-13:
When writing into EMMC through openwrt-install, select 0
enter：
meson-gxl-s905x-zte-b860h.dtb

closed 2021-04-13 by Chicadaa