ophub/amlogic-s9xxx-openwrt #54: shadowsocksplus

ylascyber, 2021-04-14:
can add shadowsocksplus on version official openwrt

ophub, 2021-04-14:
I don’t know what plug-in you are talking about. Do you have a repository address? Send me

ylascyber, 2021-04-14:
> I don’t know what plug-in you are talking about. Do you have a repository address? Send me

[image]
like this, but add on version official openwrt

ophub, 2021-04-14:
Added for you, pay attention to the compilation progress, test after completion:

https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/747903727

ramasedang, 2021-04-15:
> Added for you, pay attention to the compilation progress, test after completion:
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/747903727

so i can do same way to add passwall to lede??

ophub, 2021-04-15:
in diy-part2.sh
`svn co https://github.com/xiaorouji/openwrt-passwall/trunk package/openwrt-passwall`

in .config
```
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Shadowsocks_Server=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ShadowsocksR=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ShadowsocksR_Server=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Xray=y
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_Plus is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_GO is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Brook is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_NaiveProxy is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Kcptun is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Haproxy is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ChinaDNS_NG is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Dns2socks is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_V2ray-plugin is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Simple-obfs is not set

```

closed 2021-04-15 by ophub