ophub/amlogic-s9xxx-openwrt #59: please add LIBERNET & XDERM

nawirwii, 2021-04-20:
PLEASE ADD THE LIBERNET AND XDERM MINI INJECT TOOL IN THE NEXT UPDATE ... THANK YOU

ophub, 2021-04-20:
What is this, please tell me the specific point, send his repo address

nawirwii, 2021-04-20:
https://github.com/lutfailham96/libernet
https://github.com/ryanfauzi1

ophub, 2021-04-21:
This plug-in may not be needed by most people, so it will not be added to the default plug-in package, you can add it according to the instructions.

https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#example-1-add-a-third-party-software-package

nawirwii, 2021-04-21:
Please can't you help me how do it??
im not understand how make it please..😩🙏

ophub, 2021-04-21:
```
uci-app-frpc
luci-app-frps
luci-app-openvpn
luci-app-zerotier
```

There are already many mainstream VPN connection software in the current firmware package, so I recommend you to use these.

closed 2021-04-21 by ophub