ophub/amlogic-s9xxx-openwrt #61: I found 8189fs 5.9

Kry9toN, 2021-04-22:
I have module 8189fd for kernel 5.9
But i have problem
[image]

I dont hahave idea to fix, but i think u can give me kernel 5.9.0 or u can change localversion kernel

ophub, 2021-04-23:
8189fs for kernel 5.9 give me.
I have the kernel of `5.9.1`, I will merge it into this driver for you, please try whether it can be used normally in openwrt.

Kry9toN, 2021-04-23:
I will test it with pleasure

8189fs.ko
https://sourceforge.net/projects/krypton-project/files/OpenWrt/8189fs.ko

ophub, 2021-04-23:
I put the driver you shared into the modules-5.9.1-flippy-47+.tar.gz file, you can compile and try to see if it can drive successfully.

https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/kernel/5.9.1

Kry9toN, 2021-04-23:
I oke i try it now

Kry9toN, 2021-04-23:
> I put the driver you shared into the modules-5.9.1-flippy-47+.tar.gz file, you can compile and try to see if it can drive successfully.
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/kernel/5.9.1

[image]

Same, maybe module must be compile on same kernel

ophub, 2021-04-23:
sad.

Kry9toN, 2021-04-23:
> sad.

Maybe u can show me where you source kernel to build?

ophub, 2021-04-23:
I haven’t found this driver in the openwrt official network driver. It seems that I have to wait.

Kry9toN, 2021-04-23:
> I haven’t found this driver in the openwrt official network driver. It seems that I have to wait.

I found this module driver in khadas

https://github.com/khadas/android_hardware_wifi_realtek_drivers_8189ftv

ophub, 2021-04-23:
http://www.kernel.org/pub/linux/kernel/

ophub, 2021-04-23:
When compiling the kernel, select the network card driver and compile it in, which is the most appropriate.

Kry9toN, 2021-04-23:
> http://www.kernel.org/pub/linux/kernel/

Hmm where is makefile?

Kry9toN, 2021-04-23:
> https://www.google.com/search?q=linux+kernel+compilation
> 
> https://www.cyberciti.biz/tips/compiling-linux-kernel-26.html
> 
> https://www.linux.com/topic/desktop/how-compile-linux-kernel-0/

https://web.facebook.com/notes/dd-wrt-open-wrt-indonesia-open-source/list-download-firmware-openwrt-untuk-stb/471673447483739/

maybe you can extract this kernel for me, i try it not work on you script

ophub, 2021-04-23:
Which version of this URL do you want?

Kry9toN, 2021-04-23:
> Which version of this URL do you want?

on the link there is a list of openwrt whose wifi is working well

ophub, 2021-04-23:
Have you tried it? Which one of these can drive this network card? Connect this firmware to me, I will extract it into the kernel for you

ophub, 2021-04-23:
3.14  and 5.8.18 ?
Downloading, I'll send it to you later.

Kry9toN, 2021-04-23:
> 3.14 and 5.8.18 ?
> Downloading, I'll send it to you later.

Yes, oke thanks you, i will waiting 👍

Kry9toN, 2021-04-23:
> Have you tried it? Which one of these can drive this network card? Connect this firmware to me, I will extract it into the kernel for you

On the list everything went well with the wifi driver

ophub, 2021-04-23:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/kernel/5.8.18

You try this, the other 3.14 file cannot be mounted and cannot be used.

Kry9toN, 2021-04-23:
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/kernel/5.8.18
> 
> You try this, the other 3.14 file cannot be mounted and cannot be used.

Oke thanks i hope that work

Kry9toN, 2021-04-23:
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/kernel/5.8.18
> 
> You try this, the other 3.14 file cannot be mounted and cannot be used.

Sad, not booted

ophub, 2021-04-23:
Sad, keep working hard

closed 2021-04-23 by ophub

Kry9toN, 2021-04-23:
> Sad, keep working hard

Oke, thanks for help me