ophub/amlogic-s9xxx-openwrt #64: Amlogic S912 Wi-Fi Driver

AlexSjr, 2021-04-25:
Hi,

Could you add the Wi-Fi Driver for H96 Pro Plus (Amlogic S912 Blue PCB), please?
I believe it uses the kmod-ath10k 9377 driver but I was unable to install it because of the snapshot version.

Thanks!! :)

ophub, 2021-04-26:
These two drivers are added for you, and test them after compiling.
```
CONFIG_PACKAGE_ath10k-board-qca9377=y
CONFIG_PACKAGE_ath10k-firmware-qca9377=y
```

https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/784315724

ophub, 2021-04-26:
Can you post a photo of this device or introduction information on the Internet? Which firmware did you choose? Is it the firmware package of s912? How is it used?

AlexSjr, 2021-04-27:
Yes sure!

This is my H96 Pro Plus device with the Alfawise brand at the top (3+32GB). It has 4 USB 2.0 ports (however 2 are hidden in the case).

[image]
[image]
[image]

I chose the firmware for the S912 with kernel 5.4 to be able to install on the EMMC. So I followed the steps and selected at the time of installation the option that matches my hardware.
Product information at Gearbest: https://www.gearbest.com/tv-box-mini-pc/pp_503486.html

Thanks !!

ophub, 2021-04-27:
Thank you for your product information, I have added it to the product description form for more people to use. Thank you again for sharing.

closed 2021-04-27 by ophub

dzldd, 2021-06-01:
> 你好
> 
> 请添加 H96 Pro Plus （Amlogic S912 蓝色 PCB） 的 Wi - Fi 驱动程序吗？
> 我相信它使用kmod-ath10k 9377驱动程序，但我无法安装它，因为快照版本。
> 
> 谢谢！！:)

I am also H96 Pro Plus, which dtb file do you use?

afm175, 2022-04-21:
can you copy wireless configuration at /etc/config/wireless here?
i just installed openwrt on my S912 STB but the wireless menu in luci is gone
thanks