ophub/amlogic-s9xxx-openwrt #66: eMMC install failed when with custom build openwrt-armvirt-rootfs using local packaging method

mientz, 2021-05-12:
executing `openwrt-install` command resulting 
```
A total of [ 0 ] old partitions on EMMC will be deleted
The fdisk partition fails, the backup bootloader will be restored, and then exit.
BusyBox v1.33.0 () multi-call binary.

Usage: dd [if=FILE] [of=FILE] [ibs=N obs=N/bs=N] [count=N] [skip=N] [seek=N]
        [conv=notrunc|noerror|sync|fsync]
        [iflag=skip_bytes|fullblock|direct] [oflag=seek_bytes|append|direct]

Copy a file with converting and formatting

        if=FILE         Read from FILE instead of stdin
        of=FILE         Write to FILE instead of stdout
        bs=N            Read and write N bytes at a time
        ibs=N           Read N bytes at a time
        obs=N           Write N bytes at a time
        count=N         Copy only N input blocks
        skip=N          Skip N input blocks
        seek=N          Skip N output blocks
        conv=notrunc    Don't truncate output file
        conv=noerror    Continue after read errors
        conv=sync       Pad blocks with zeros
        conv=fsync      Physically write data out before finishing
        conv=swab       Swap every pair of bytes
        iflag=skip_bytes        skip=N is in bytes
        oflag=seek_bytes        seek=N is in bytes
        iflag=direct    O_DIRECT input
        oflag=direct    O_DIRECT output
        iflag=fullblock Read full blocks
        oflag=append    Open output in append mode

N may be suffixed by c (1), w (2), b (512), kB (1000), k (1024), MB, M, GB, G
```
Details 

- packaging method : [local packaging](https://github.com/ophub/amlogic-s9xxx-openwrt#local-packaging-instructions)
- armvirt-rootfs file : **custom build** `openwrt-21.02-rc1` openwrt-armvirt-64-default-rootfs.tar.gz
- device : hg680p - using default `meson-gxl-s905x-p212.dtb`

note: i also build your `install-program` source when build the custom rootfs img

mientz, 2021-05-12:
is there some dependecy that needed to be installed, or some partitioning before executing the command that you already include on your `openwrt-armvirt-64-default-rootfs` build that missing from my build? 

Thanks

ophub, 2021-05-12:
Download one of my `openwrt-armvirt-64-default-rootfs.tar.gz` for local packaging and see if it can be written to emmc, so that the problem of the packaging environment can be ruled out first, and the compiled dependent packages can be found.

ophub, 2021-05-12:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/1223222aa959dbde1c80a7862bb9adf827475989/router-config/lede-master/.config#L541

From line 541 to the end, there are several dependencies related to installation. It's basically the same in the 21.02 branch.

mientz, 2021-05-12:
> Download one of my `openwrt-armvirt-64-default-rootfs.tar.gz` for local packaging and see if it can be written to emmc, so that the problem of the packaging environment can be ruled out first, and the compiled dependent packages can be found.

Already done it before and its working with your `rootfs` build.

> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/1223222aa959dbde1c80a7862bb9adf827475989/router-config/lede-master/.config#L541
> 
> From line 541 to the end, there are several dependencies related to installation. It's basically the same in the 21.02 branch.

This, is what i need. Thankyou! 👍 . I remove some packages from your default `.config` because i didn't all of those feature, and i unknowingly remove some necessary packages.

closed 2021-05-12 by mientz