ophub/amlogic-s9xxx-openwrt #73: 大佬，有没有s912的H96pro+的dtb文件，找到CoreELEC用的dtb, gxm_q201_3g_1gbit，不知能通用否。

dzldd, 2021-05-30:
大佬，有没有s912的H96pro+的dtb文件，找到CoreELEC用的dtb, gxm_q201_3g_1gbit，不知能通用否。

ophub, 2021-05-30:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/amlogic-s9xxx/amlogic-dtb/meson-gxm-octopus-planet.dtb

s912 universal

dzldd, 2021-05-30:
谢谢大佬，这是octopus的，也是目前在用的，因为好多信息都显示不出来，适配不是很完美，就想问问有没有H96pro+自已的dtb文件，能适配的完美些

ophub, 2021-05-30:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/amlogic-s9xxx/amlogic-dtb/meson-gxm-q200.dtb

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/amlogic-s9xxx/amlogic-dtb/meson-gxm-q201.dtb

dzldd, 2021-05-30:
谢谢大佬

closed 2021-05-30 by ophub

dzldd, 2021-05-31:
大佬，请问已经从U盘刷入EMMC了，能否从EMMC启动的Openwrt系统中更换dtb文件，还是必须重新刷机？谢谢！

dzldd, 2021-05-31:
看到大佬为之前有个老外S912添加编译了无线驱动

CONFIG_PACKAGE_ath10k-board-qca9377=y
CONFIG_PACKAGE_ath10k-firmware-qca9377=y
CONFIG_PACKAGE_kmod-ath10k=y
我的盒子和他的是一样的，请问现在的固件里还有这些驱动吗？因为我的无线现在也还没有启动起来，谢谢大佬！