ophub/amlogic-s9xxx-openwrt #74: 大佬，请问已经从U盘刷入EMMC了，能否从EMMC启动的Openwrt系统中更换dtb文件，还是必须重新刷机？谢谢！

dzldd, 2021-06-01:
大佬，请问已经从U盘刷入EMMC了，能否从EMMC启动的Openwrt系统中更换dtb文件，还是必须重新刷机？看到大佬为之前为有个老外S912添加编译了无线驱动

CONFIG_PACKAGE_ath10k-board-qca9377=y
CONFIG_PACKAGE_ath10k-firmware-qca9377=y
CONFIG_PACKAGE_kmod-ath10k=y
我的盒子和他的是一样的，请问现在的固件里还有这些驱动吗？因为我的无线现在也还没有启动起来，谢谢大佬！

dzldd, 2021-06-01:
看到大佬为之前为有个老外S912添加编译了无线驱动，我的盒子和他的是一样的，

CONFIG_PACKAGE_ath10k-board-qca9377=y
CONFIG_PACKAGE_ath10k-firmware-qca9377=y
CONFIG_PACKAGE_kmod-ath10k=y
请问现在的固件里还有这些驱动吗？因为我的无线现在也还没有启动起来，谢谢大佬！

ophub, 2021-06-01:
yes, you can
Login in to openwrt → system menu → Amlogic Service → Replace OpenWrt Kernel

ophub, 2021-06-01:
Today's new version supports manual upload of firmware or kernel for upgrade, as well as direct online download of firmware and kernel package upgrade.

Login in to openwrt → system menu → Amlogic Service

https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/894545875

Under compiling, then download and use, or you can directly upgrade from the existing firmware

ophub, 2021-06-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/openwrt-master/.config

The driver is included here, you can search it, you can fork and then you can add, delete and modify by yourself

dzldd, 2021-06-01:
我看你以的链接.config里CONFIG_PACKAGE_ath10k-board-qca9377=y，CONFIG_PACKAGE_ath10k-firmware-qca9377=y，CONFIG_PACKAGE_kmod-ath10k=y 都是选中了的，是不是今天这个固件里就有这个9377的无线网卡驱动，谢谢！

ophub, 2021-06-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/895145023

Wait for this compilation to download and test

dzldd, 2021-06-01:
thank you very much

ophub, 2021-06-01:
download from here:

https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/openwrt_s9xxx_lede_2021.06.01.1808

dzldd, 2021-06-02:
上传固件后    通过终端 → 输入升级命令: openwrt-update，显示升级成功，但是重启没能成功，重启后未能接入网络，明天通过U盘重刷再试。谢谢！

ophub, 2021-06-02:
The version span is large, do not keep the configuration. If `reboot` fails to restart, unplug the power supply and plug in the power supply to start.

dzldd, 2021-06-02:
我升级前的版本是5.4.118，保留配置升级的，拨掉电源，重启未能成功，准备晚上用U盘重刷，直接用U盘重刷可以的吧，不用重新刷入原来的TV系统再重刷？

dzldd, 2021-06-02:
谢谢，就是能用最新编译的固件，对吧，也能直接用U盘刷，是吧？

ophub, 2021-06-02:
yes

dzldd, 2021-06-02:
谢谢，但是无线还是不能用，只能用章鱼的dtb文件，昨天别外两个dtb刷了启动不了。

ophub, 2021-06-02:
test this:

https://github.com/ophub/op/releases/tag/openwrt_s9xxx_2021.06.03.0706

dzldd, 2021-06-03:
好的，我再测试下，昨天，也就是2号的固件刷入后恢复备份存在密码错误的问题，前天1号的固件刷入后恢复备份存在网络接口配置页面死循环的问题，也许是我的个例，今天再试下这个最新的，谢谢ophub一直的支持。

ophub, 2021-06-03:
https://github.com/openwrt/luci/commits/master

In the past week, the openwrt official website is changing the file information in `/etc/config/network`, changing the node name from ifname to device.

At present, the official docker network node has not been modified, so it will cause unlimited pop-ups on the network interface. I have added a correction command to the startup task. You can reset the pop-up window twice.

closed 2021-06-04 by ophub