ophub/amlogic-s9xxx-openwrt #78: 2/16 version

aska-lz, 2021-06-13:
hi develover, kernel 5.12.10 booting normaly in x96max+ ver.4/32 . 
but can't booting on x96max+ ver.2/16 . 
please fix it.
thank you 🙏

ophub, 2021-06-13:
Hello, 2G version and 4G version are two different firmwares, 2G needs to use s905x2 firmware.

E.g: openwrt_s905x2_v5.12.10_2021.06.13.0352.img.gz

aska-lz, 2021-06-13:
> Hello, 2G version and 4G version are two different firmwares, 2G needs to use `s905x2` firmware.
> 
> E.g: `openwrt_s905x2_v5.12.10_2021.06.13.0352.img.gz`
> 
> Use the openwrt-install command to install option `10. X96-Max-2G -------------- [ S905x2 / CPU: 1944Mtz / NETWORK: 100M ]`
> 
> Use the luci-app-amlogic install option:
> 
> [image]

oh, i think same come with s905x3 , thank you for your reply. i will try with  s905x2

closed 2021-06-13 by aska-lz