ophub/amlogic-s9xxx-openwrt #79: cannot find dependency kernel

aska-lz, 2021-06-19:
Installing stunnel (5.59-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/stunnel_5.59-1_aarch64_cortex-a53.ipk
Configuring stunnel.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-crypto-null
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-crypto-aead
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-crypto-manager
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-crypto-authenc

ophub, 2021-06-19:
This ipk cannot be installed directly, and the relevant dependencies need to be compiled together when the firmware is compiled. You can compile after adding related software packages and dependencies according to your needs.

This ipk is suitable for 5.4 openwrt firmware.

aska-lz, 2021-06-19:
> This ipk cannot be installed directly, and the relevant dependencies need to be compiled together when the firmware is compiled. You can compile after adding related software packages and dependencies according to your needs.
> 
> This ipk is suitable for 5.4 openwrt firmware.

Installing vpnbypass (1.3.1-10) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/vpnbypass_1.3.1-10_all.ipk
Configuring vpnbypass.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-c3cda308942512b5d47916e0bc2a8a17) for kmod-ipt-ipset

ophub, 2021-06-19:
This ipk cannot be installed directly, and the relevant dependencies need to be compiled together when the firmware is compiled.

aska-lz, 2021-06-19:
ok, waiting for next update

closed 2021-06-19 by aska-lz