ophub/amlogic-s9xxx-openwrt #95: badvpn-tun2socks

lynxnexy, 2021-06-30:
please add package "badvpn-tun2socks" to system firmware

codesnas, 2021-07-01:
Hello, I did not find the package you mentioned. What is his full name.

ophub, 2021-07-01:
Hello, I did not find the package you mentioned. What is his full name.

kbuku35, 2021-07-01:
maybe this means :
https://github.com/zhfish/tun2socks-Openwrt

ophub, 2021-07-01:
https://github.com/zhfish/tun2socks-Openwrt/blob/master/badvpn/Makefile
```
PKG_NAME:=badvpn
PKG_VERSION:=1.999.117rc2
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.bz2
PKG_SOURCE_URL:=http://badvpn.googlecode.com/files/
```

The author hasn't updated it for 8 years. This file can no longer be downloaded. Give up, face the future, spring blossoms.

lynxnexy, 2021-07-01:
this is https://code.google.com/archive/p/badvpn/wikis/tun2socks.wiki

i need this package for ssh tunneling ==> dynamic socks5 ==> tun interface

ophub, 2021-07-02:
His file can no longer be downloaded, so give up.

closed 2021-07-02 by ophub

helmiau, 2021-08-10:
> Find another plan.

how about this repo?
https://github.com/friyadhibiermann/openwrt_badvpn-tun2socks