ophub/amlogic-s9xxx-openwrt #96: Add interface on system fw

riyuejz, 2021-07-03:
can you help add the hilink modem interface to the system firmware

> Interface setup details

Name `hilink`
Protocol `dhcp client`
Physical `eth1`
Firewall `wan`

ophub, 2021-07-03:
On your device: `Network` > `Interface`, after the configuration is completed, you can get your dedicated `network` file. The path is in `/etc/config/network` of the `openwrt` system.

`Fork` the repo, copy this file to the `/amlogic-s9xxx/common-files/files/etc/config/network` path in the repo. In this way, your packaged `openwrt` firmware is already set.

riyuejz, 2021-07-04:
For example, if I put a file in `/usr/bin/mybinnary` will the file automatically give permission to 755?

ophub, 2021-07-04:
There are two ways to authorize. The first method is to locally `chmod +x usr/bin/mybinnary` and then `git push` to github. Another method is to add permissions when packaging, such as adding `chmod +x usr/bin/mybinnary` in `make`

riyuejz, 2021-07-04:
ok i will try it .. thanks and good morning 😄

closed 2021-07-04 by ophub