ophub/amlogic-s9xxx-openwrt #97: Wifi Issue x96 max +

johnmalay, 2021-07-06:
i've have issue of the wifi network for x96 max +. the network interface shows nothing in openwrt wireless setting. i'm referring this first issue i create https://github.com/ophub/amlogic-s9xxx-openwrt/issues/24

after finishing install openwrt to the stb, opening network setting and no wifi adapter. seems like driver for the stb is not install. is there any update for the latest build that fix the issue of wifi?

ophub, 2021-07-06:
The answer to the question about the wifi module of X96-Max+: X96-Max+ uses the RTL8822CS (SDIO) module. Officially released the first version of the driver (rtw88) in 2019. At that time, it only supported PCIE RTL88xxXE, but it also stated that it will support it in the future. RTL88xxXU of USB and RTL88xxXS of SDIO, but until 2021, no follow-up drivers have been seen.

https://www.phoronix.com/scan.php?page=news_item&px=Realtek-RTW88-WiFi-Linux-Driver

So as of now, the wifi of X96-Max+ cannot be used.

johnmalay, 2021-07-07:
ok thanks. i should buy another box. last question,  is there any wifi issue with hk1 box? this box [https://shopee.com.my/HK1-BOX-Android-9.0-Smart-TV-BOX-4GB-64GB-128GB-Amlogic-S905X3-2.4G-5G-Wifi-1000M-BT-8K-IPTV-Netflix-Youtube-Set-Top-Box-i.230876871.5630734452](url)

ophub, 2021-07-07:
HK-1 box's wireless can be used normally.

johnmalay, 2021-07-07:
Tqvm for the reply. 👍

closed 2021-07-07 by johnmalay