ophub/amlogic-s9xxx-openwrt #98: modem info

riyuejz, 2021-07-06:
hi can you help me to install this package into the firmware i tried to manually install .ipk but failed

package:
`https://github.com/4IceG/luci-app-3ginfo`

ophub, 2021-07-06:
- in https://github.com/ophub/amlogic-s9xxx-openwrt/blob/32ab401fb17256317a4d23da82250c0f54d7d909/router-config/lede-master/diy-part2.sh#L46

`svn co https://github.com/koshev-msk/luci-app-modeminfo/trunk package/luci-app-modeminfo`

- in https://github.com/ophub/amlogic-s9xxx-openwrt/blob/32ab401fb17256317a4d23da82250c0f54d7d909/router-config/lede-master/.config#L536

`CONFIG_PACKAGE_luci-app-modeminfo=y`

riyuejz, 2021-07-06:
ok i will try, thanks for the help

closed 2021-07-07 by ophub