ophub/amlogic-s9xxx-openwrt #99: 能否支持A311D？

yangquan3, 2021-07-17:


ophub, 2021-07-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/amlogic-s9xxx/amlogic-dtb/meson-g12b-a311d-khadas-vim3.dtb

`/boot/uEnv.txt` 
FDT changed to the file name of `meson-g12b-a311d-khadas-vim3.dtb`.

Open the A311D-Box in ADB and start the USB flash drive:

adb connect device_ip_address
adb shell reboot update

Test whether you can start the openwrt system in the U disk.

closed 2021-07-19 by ophub

darkliang, 2022-02-12:
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/amlogic-s9xxx/amlogic-dtb/meson-g12b-a311d-khadas-vim3.dtb
>
> `/boot/uEnv.txt` FDT changed to the file name of [...]

Hi, I have a A311D box (Khadas vim3) also. I used the S922X image and replaced the FDT to `meson-g12b-a311d-khadas-vim3.dtb`, it works. But it doesn't support the wireless drivers, and it can't turn on the fan. So how can I build a custom-made image for it?

ophub, 2022-02-12:
https://github.com/armbian/build/blob/master/config/boards/khadas-vim3.conf

I saw this box in the official support list of armbian. The official default compilation should be able to support it, right?
The boxes in my repository are devices that are not officially supported by default, and we tweaked a little bit to try and make armbian work on these boxes.

darkliang, 2022-02-12:
I'm sorry being new to this, I'm not sure what Armbian support has to do with OpenWRT.

ophub, 2022-02-12:
You can think of it as the same. Two different systems, but a twin brother's system for the box

Ravjhay, 2025-01-05:
how to hack scatter game