ophub/amlogic-s9xxx-openwrt #102: Install to emmc failed

kyliezero, 2021-07-26:
can't install to emmc, even though I'm currently booting the system via the sdcard..
[image]

ophub, 2021-07-26:
try use usb

ophub, 2021-07-26:
- I just updated the plug-in and fixed the installation judgment.
- Please set the network-lan first after the usb is started, fill in the gateway, and ensure that the box can access github.com online.
- Run the following command to update the plugin to the latest version, you can install it normally.

`curl -fsSL git.io/luci-app-amlogic | bash`

Or you can download the latest version in the [Release](https://github.com/ophub/amlogic-s9xxx-openwrt/releases) tomorrow, and the new version will be automatically compiled soon.

PascalDnZO, 2021-07-27:
I have the problem, can you fix it

1. instaling status is done
[image]

2. after unpluging usb disk , i can't access luci to reboot the device 
[image]

3. when i check the terminal to reboot the device
[image]

status terminal "segmentation fault"

4. after I unplug and plug in the adapter
[image]

ophub, 2021-07-27:
1. Install, prompting that the installation is complete.
2. Unplug the usb, unplug the power
3. Plug in the power and restart

PascalDnZO, 2021-07-27:
Ok , i success install to emmc , thanks
1. Install , prompting that the installation is complete.
2. Unplug power , re-plug power
3. After booting , unplug usb disk 
4. Reboot

if after 90 seconds the computer does not get a connection, then unplug it and plug in the power

ophub, 2021-07-27:
After the installation is successful, you can remove the usb and plug in the power again to let openwrt start from emmc. No need to wait.

kyliezero, 2021-07-28:
my device can't boot via usb cardreder, it can only go through the sdcard slot, is there a solution so I can install fw to emmc via the sdcard slot?

kyliezero, 2021-07-28:
can you revert the install method from terminal 'openwrt-install'

ophub, 2021-07-28:
The previous commands can be used, such as
```
Installation: openwrt-install
Update: openwrt-update
Backup: openwrt-backup
```

Luci also works through these commands.

kyliezero, 2021-07-28:
Ok, siap

closed 2021-07-28 by ophub

kyliezero, 2021-07-28:
$ ssh root@192.168.1.1
root@192.168.1.1's password:

BusyBox v1.33.1 (2021-07-28 05:41:38 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r0-3c6aadc
 -----------------------------------------------------
 Install: OpenWrt → System → Amlogic Service → Install
 Amlogic SoC: s905x
 OpenWrt Kernel: 5.4.133-flippy-62+o
 Packaged Date: 2021-07-28
 -----------------------------------------------------
root@LYNX:~# openwrt-install
AUTO_MAINLINE_UBOOT:
SPECIFY_SOC:
SPECIFY_DTB:
/dev/mmcblk1 mmcblk2
grep: mmcblk2: No such file or directory
mmcblk1
mmcblk2 :  bytes
ROOTFS: mmcblk1p2
BOOT: mmcblk1p1
------------------------------------------------------------------------
Please select s9xxx box model:
1. X96-Max+ (4G DDR) -------- [ S905x3 / CPU: 2124Mtz / NETWORK: 1000M ]
2. X96-Max+ (4G DDR) -------- [ S905x3 / CPU: 2208Mtz / NETWORK: 1000M ]
3. HK1-Box (4G DDR) --------- [ S905x3 / CPU: 2124Mtz / NETWORK: 1000M ]
4. HK1-Box (4G DDR) --------- [ S905x3 / CPU: 2184Mtz / NETWORK: 1000M ]
5. H96-Max-X3 (4G DDR) ------ [ S905x3 / CPU: 2124Mtz / NETWORK: 1000M ]
6. H96-Max-X3 (4G DDR) ------ [ S905x3 / CPU: 2208Mtz / NETWORK: 1000M ]
7. Ugoos-X3 (Cube/Pro/Plus) - [ S905x3 / CPU: 2124Mtz / NETWORK: 1000M ]
8. Ugoos-X3 (Cube/Pro/Plus) - [ S905x3 / CPU: 2208Mtz / NETWORK: 1000M ]
9. X96-Max-4G --------------- [ S905x2 / CPU: 1944Mtz / NETWORK: 1000M ]
10. X96-Max-2G -------------- [ S905x2 / CPU: 1944Mtz / NETWORK: 100M  ]
11. Belink-GT-King ---------- [ S922x  / CPU: 2124Mtz / NETWORK: 1000M ]
12. Belink-GT-King-Pro ------ [ S922x  / CPU: 2124Mtz / NETWORK: 1000M ]
13. UGOOS-AM6-Plus ---------- [ S922x  / CPU: 2124Mtz / NETWORK: 1000M ]
14. Octopus-Planet ---------- [ S912   / CPU: 2124Mtz / NETWORK: 1000M ]
15. Phicomm-n1 -------------- [ S905d  / CPU: 2124Mtz / NETWORK: 1000M ]
16. hg680p & b860h ---------- [ S905x  / CPU: 2124Mtz / NETWORK: 1000M ]

0. Other ---------------- [ Enter the dtb file name of your box ]
------------------------------------------------------------------------
Please choose:
16
FDT Value [ meson-gxl-s905x-p212.dtb ]
This Kernel [ 5.4.133-flippy-62+o ]
Backup bootloader -> [ BackupOldBootloader.img ] ...
BusyBox v1.33.1 (2021-07-28 05:41:38 UTC) multi-call binary.

Usage: dd [if=FILE] [of=FILE] [ibs=N obs=N/bs=N] [count=N] [skip=N] [seek=N]
        [conv=notrunc|noerror|sync|fsync]
        [iflag=skip_bytes|fullblock|direct] [oflag=seek_bytes|append|direct]

Copy a file with converting and formatting

        if=FILE         Read from FILE instead of stdin
        of=FILE         Write to FILE instead of stdout
        bs=N            Read and write N bytes at a time
        ibs=N           Read N bytes at a time
        obs=N           Write N bytes at a time
        count=N         Copy only N input blocks
        skip=N          Skip N input blocks
        seek=N          Skip N output blocks
[... 55 lines cut ...]

Copy a file with converting and formatting

        if=FILE         Read from FILE instead of stdin
        of=FILE         Write to FILE instead of stdout
        bs=N            Read and write N bytes at a time
        ibs=N           Read N bytes at a time
        obs=N           Write N bytes at a time
        count=N         Copy only N input blocks
        skip=N          Skip N input blocks
        seek=N          Skip N output blocks
        conv=notrunc    Don't truncate output file
        conv=noerror    Continue after read errors
        conv=sync       Pad blocks with zeros
        conv=fsync      Physically write data out before finishing
        conv=swab       Swap every pair of bytes
        iflag=skip_bytes        skip=N is in bytes
        oflag=seek_bytes        seek=N is in bytes
        iflag=direct    O_DIRECT input
        oflag=direct    O_DIRECT output
        iflag=fullblock Read full blocks
        oflag=append    Open output in append mode

N may be suffixed by c (1), w (2), b (512), kB (1000), k (1024), MB, M, GB, G
root@LYNX:~#

kyliezero, 2021-07-28:
root@LYNX:~# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda            8:0    1    0B  0 disk
mmcblk1      179:0    0  7.4G  0 disk
├─mmcblk1p1  179:1    0  243M  0 part /boot
└─mmcblk1p2  179:2    0    1G  0 part /
mmcblk2      179:32   0  7.3G  0 disk
├─mmcblk2p1  179:33   0  160M  0 part /mnt/mmcblk2p1
├─mmcblk2p2  179:34   0    1G  0 part /mnt/mmcblk2p2
├─mmcblk2p3  179:35   0    1G  0 part /mnt/mmcblk2p3
└─mmcblk2p4  179:36   0  4.4G  0 part /mnt/mmcblk2p4
mmcblk2boot0 179:64   0    4M  1 disk
mmcblk2boot1 179:96   0    4M  1 disk
root@LYNX:~#