ophub/amlogic-s9xxx-openwrt #108: Local Compile Guide

odinotisan, 2021-08-22:
How do we include packages like `diy-part2.sh` in local compilation?

ophub, 2021-08-23:
In the source directory where openwrt is compiled, enter the path where the diy2.sh script is located and execute it, for example:

```
 ├── router                                           
     ├── amlogic-s9xxx-openwrt           # https://github.com/ophub/amlogic-s9xxx-openwrt related code files
     │   ├── router-config
     │       ├── lede-master
     │          ├── diy-part1.sh
     │          └── diy-part2.sh
     │
     ├── openwrt                # https://github.com/coolsnowwolf/lede related code files
     │   ├── bin
     │   ├── package
     │   └── feeds

cd router/openwrt
../amlogic-s9xxx-openwrt/router-config/lede-master/diy-part2.sh
```
amlogic-s9xxx-openwrt

odinotisan, 2021-08-23:
Thank you @ophub 

Do you face this issue `router/openwrt/include/toplevel.mk:230: world] Error 2` 

```
ifeq ($(SDK),1)

%::
        @+$(PREP_MK) $(NO_TRACE_MAKE) -r -s prereq
        @./scripts/config/conf $(KCONF_FLAGS) --defconfig=.config Config.in
        @+$(ULIMIT_FIX) $(SUBMAKE) -r $@

else

%::
        @+$(PREP_MK) $(NO_TRACE_MAKE) -r -s prereq
        @( \
                cp .config tmp/.config; \  #### <====== ISSUE LINE 230
                ./scripts/config/conf $(KCONF_FLAGS) --defconfig=tmp/.config -w tmp/.config Config.in > /dev/null 2>&1; \
                if ./scripts/kconfig.pl '>' .config tmp/.config | grep -q CONFIG; then \
                        printf "$(_R)WARNING: your configuration is out of sync. Please run make menuconfig, oldconfig or defconfig!$(_N)\n" >&2; \
                fi \
        )
        @+$(ULIMIT_FIX) $(SUBMAKE) -r $@ $(if $(WARN_PARALLEL_ERROR), || { \
                printf "$(_R)Build failed - please re-run with -j1 to see the real error message$(_N)\n" >&2; \
                false; \
        } )

endif
```

ophub, 2021-08-23:
Compile locally for the first time, set:
`make -j1 V=s`

odinotisan, 2021-08-23:
I figured, it was due to running as root

`export FORCE_UNSAFE_CONFIGURE=1`

Here's the fix, who know it might help someone.

Was figuring how to load the config by reading your workflows. It happens you posted! Thank you so much!

Was trying to build for rtl8812au-ac. Hopefully it works.

ophub, 2021-08-23:
Compilation of openwrt files, `root` user cannot be used.

https://github.com/coolsnowwolf/lede/blob/master/README_EN.md

odinotisan, 2021-08-23:
Btw, the main readme for S905x2 has a tokopedia link to x96 mini instead of x96 max. Which doesn't work. In order for x96 mini to work, you have to use s905x and change the dtb file to s905w of tx3 mini. It works and boot up perfectly.

odinotisan, 2021-08-23:
> I figured, it was due to running as root
> 
> `export FORCE_UNSAFE_CONFIGURE=1`
> 
> Here's the fix, who know it might help someone.
> 
> Was figuring how to load the config by reading your workflows. It happens you posted! Thank you so much!
> 
> Was trying to build for rtl8812au-ac. Hopefully it works.

> Compilation of openwrt files, `root` user cannot be used.
> 
> https://github.com/coolsnowwolf/lede/blob/master/README_EN.md
> 
> ```
> Note:
> DO NOT USE root USER FOR COMPILING!!!
> ```

I missed that. Thanks!

ophub, 2021-08-23:
> Btw, the main readme for S905x2 has a tokopedia link to x96 mini instead of x96 max. Which doesn't work. In order for x96 mini to work, you have to use s905x and change the dtb file to s905w of tx3 mini. It works and boot up perfectly.

Thank you very much for the information, I have listed s905w separately.

odinotisan, 2021-08-23:
It seems to be overriding the `.config` I have copied from this repo. Whenever `make desconfig` ran. Following the workflows action in this repo. It shows that it needs to run `make desconfig` instead of `make menuconfig`. 

Any clue? I have this USB Wireless Adapter I am attempting to compile the module. TPLink T2U NANO.

http://en.techinfodepot.shoutwiki.com/wiki/TP-LINK_Archer_T2U_Nano

On the link above it shows RTL8811AU.

But when I did `lsusb` it show the same as RTL8821AU

Hence, I tried this module but does not work.
https://github.com/aircrack-ng/rtl8812au

Then I read on openwrt forum. Someone made a patch. Read last few comments.
https://forum.openwrt.org/t/rtl8812au-and-or-rtl8814au-drivers/67/193

Thus, I have created a repo for `diy-part2.sh`
https://github.com/odinotisan/rtl8812au-ac

But it's always not included in the build. Please help.

Does the lede make only allows 1core? It tooks to long me to try things out, that's also the reason I move from github actions to local compilation.

Any help is appreciated, thank you.

ophub, 2021-08-23:
- The driver is located in the `/lib/modules/*/kernel` directory
- When packaging openwrt, it has been replaced by `modules-*.tar.gz` in the kernel package:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/5786d6f532f6ff1fe416bfeab000dc51732d08bb/make#L135

So you need to compile a kernel that contains the driver you want, and then package it into openwrt. E.g:

https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/5.4.132

odinotisan, 2021-08-23:
>     * The driver is located in the `/lib/modules/*/kernel` directory
> 
>     * When packaging openwrt, it has been replaced by `modules-*.tar.gz` in the kernel package:
>       https://github.com/ophub/amlogic-s9xxx-openwrt/blob/5786d6f532f6ff1fe416bfeab000dc51732d08bb/make#L135
> 
> 
> So you need to compile a kernel that contains the driver you want, and then package it into openwrt. E.g:
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-kernel/5.4.132
> 
>     * Download from the official website of the kernel and compile a kernel package that can be used by `amlogic arm64`. https://www.kernel.org/
> 
>     * It is recommended that you start the `armbian` system on the current device for packaging. Or cross-compile on x86_64 devices. Can you compile the kernel?

I was about to tell you that x96 mini does not work with 5.12 or 5.13. I tried all the kernel provided through this repo. About to give up. It only works with 5.4.xx

ophub, 2021-08-23:
Kernels 5.10 and above need the corresponding mainline u-boot support. There is no s905w at present, wait for flippy to provide it

ophub, 2021-08-24:
I saw in the armbian community that someone installed the armbian system in the TX3 of s905w, I copied the u-boot file from this system, you try this, it is the same as x96-mini.

Installing Armbian on Amlogic S905w (Tanix TX3 Mini)

https://i12bretro.github.io/tutorials/0316.html

https://www.youtube.com/watch?v=n54aEMdXj7I

Since I don’t have s905w equipment, I can’t test it. Please feedback the result after testing. If this is not possible, I will look for it again. If you find a u-boot file of s905w that can be supported, please share it, thank you.

odinotisan, 2021-08-24:
Thank you! I will give it a spin and update you! Thanks your hardwork!