ophub/amlogic-s9xxx-openwrt #111: Local source compile error

johnmalay, 2021-09-05:
i'm getting this error while compiling firmware base on s905x3

(1.1) Start packaging OpenWrt [ s905x3 - 5.14.1 ]. Remaining space is 339G.
 [  s905x3  -  5.14.1  ]  (1/6) extract armvirt files.
 [  s905x3  -  5.14.1  ]  (2/6) extract armbian files.
 [  s905x3  -  5.14.1  ]  (3/6) refactor related files.
 [  s905x3  -  5.14.1  ]  (4/6) make openwrt image.
 [  s905x3  -  5.14.1  ]  (5/6) copy files to image.
mount: /home/xx/amlogic-s9xxx-openwrt/tmp/mount/5.14.1/s905x3/rootfs: wrong fs type, bad option, bad superblock on /dev/loop5p2, missing codepage or helper program, or other error.
 [  Error  ] mount /dev/loop5p2 failed!

what is the root cause of this error?

johnmalay, 2021-09-05:
i'm testing direct command without using selection guide from the script.
sudo ./make -d -b s905x3 -k 5.4.144. (1.1) OpenWrt packaged successfully.  and no error occured. 
just wonder what is the main cause of error above. i've notice when selecting number (4) s905w soc, it randomly select another soc (1)s905x3. the script automatically choose for me.

ophub, 2021-09-05:
Before local compilation and packaging, first install the environment dependencies.

sudo apt-get update -y
sudo apt-get full-upgrade -y
sudo apt-get install -y $(curl -fsSL git.io/ubuntu-2004-openwrt)

johnmalay, 2021-09-05:
tq issue solve.

ophub, 2021-09-05:
ok

closed 2021-09-05 by ophub