ophub/amlogic-s9xxx-openwrt #112: luci app and so many others tools removed

latiefkhan, 2021-09-06:
why somay luci app and so many others tools removed .?
like openclash.?
and others tools

ophub, 2021-09-06:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/caa6443c8c7154e278185912456237bb8b844f5c

I removed some packages that most people might not need. If you need, you can fork it, add the required software packages, and compile your own personalized openwrt.

closed 2021-09-07 by ophub

johnmalay, 2021-09-10:
help me how to clone this repo in github action.
https://github.com/helmiau/helmiwrt-packages.git
i've already set the url inside diy part 2. and insert require config file in .config after build using action the package is not found. I try to local build and I can see the package when build. what is the correct url or tehcnique to pull the package using action.