ophub/amlogic-s9xxx-openwrt #116: X96 Mini No Wifi

LeosSire, 2021-09-09:
Following the table in the readme I purchased an X96 mini but there is no wifi.

Is there a list of tested units which have working wifi?

The wifi chip for the X96-mini is "sv605ip" of which I can find very little online.

I've also seen mention of the X96-max not having working wifi on linux systems but I believe wifi works using LibreElec (used it 6+ months ago).

ophub, 2021-09-09:
The wireless network card used by x96, currently openwrt does not have its driver. hk1, ugoos, belink boxes can use wifi.

closed 2021-09-10 by ophub

LeosSire, 2021-09-11:
ophub, thank you for your feedback, I've been looking for a hardware solution for a long time to purchase a decent number of boxes for flashing a custom version of OpenWRT onto. 

The UGOOS and Beelink (branded Beelink) boxes are all circa $100+ the HK1Box is the only reasonably priced version. Or [and this is a big or] are boxes like the low end TrongleS3 (Mini MXIII II made by beelink) likely to work? Or are the low cost Beelink's the traditional hoover up cheap parts and throw them onto a board? My stock low cost TrongleS3 (board made by Beelink) has an Aigale AW-CM273SM which uses a Atheros QCA9377 chip (which is included on the router-config/lede-master/.config file). 

What are your thoughts on android being able to incorporate the wifi drivers on low cost boxes but no one can, and by no one, I mean the other 3rd party linux dev's offering Armbian etc... not able to incorporate it. Custom android rom's seem to do okay. Bit weird.

ophub, 2021-10-02:
s903x3 hk1 box can use wifi.