ophub/amlogic-s9xxx-openwrt #117: 3rd party repo

kokenot, 2021-09-10:
help me how to clone this repo in github action.
https://github.com/helmiau/helmiwrt-packages.git
i've already set the url inside diy part 2. and insert require config file in .config after build using action the package is not found. I try to local build and I can see the package when build. what is the correct url or technique to pull the package using action.

ophub, 2021-09-10:
git clone https://github.com/helmiau/helmiwrt-packages.git package/helmiwrt-packages

ophub, 2021-09-10:
In the source directory where `openwrt` is compiled, enter the path where the `diy-part2.sh` script is located and execute it, for example:

```
 ├── router                                           
     ├── amlogic-s9xxx-openwrt     # https://github.com/ophub/amlogic-s9xxx-openwrt related code files
     │   ├── router-config
     │       ├── lede-master
     │          ├── diy-part1.sh
     │          └── diy-part2.sh
     │
     ├── openwrt     # https://github.com/coolsnowwolf/lede related code files
     │   ├── bin
     │   ├── package
     │   └── feeds

cd router/openwrt     # Enter the openwrt directory
../amlogic-s9xxx-openwrt/router-config/lede-master/diy-part2.sh       #Enter the relative path of diy-part2.sh and press Enter to execute
```

ophub, 2021-09-10:
for more:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md

kokenot, 2021-09-10:
tq. solve

closed 2021-09-11 by ophub