ophub/amlogic-s9xxx-openwrt #119: error compiling 3rd party repo

kokenot, 2021-09-11:
i tried several time compiling 3rd party repo but failed. already done what u said at this post #117. ssh then run the command to pull package inside container and it 'success' but after continue compiling the firmware i've got an error. 
here is the error
[image]

ophub, 2021-09-11:
Two ways:
1. Local `make menuconfig` gets the complete `.config` and replaces the files in the warehouse.

2. Manually change `.config` to change gradually. Add a small amount each time, and gradually change the fine-tuning.

kokenot, 2021-09-11:
diy-part2.sh must be run again inside container after I replace the config?

ophub, 2021-09-12:
The first execution of diy-part1.sh and diy-part2.sh is to `git clone` the plug-in introduced in this file to the local.

kokenot, 2021-09-12:
ok i will setup the local environment first and let you know the result

kokenot, 2021-09-12:
issue solve. compiling on local machine.

ophub, 2021-09-12:
Congratulations. The complete .config file generated locally can overwrite the simplified .config file on github for compilation on github.

closed 2021-09-12 by ophub