ophub/amlogic-s9xxx-openwrt #120: 最近固件，N1盒子无法安装iptables-mod-tproxy

searchingx, 2021-09-16:
N1盒子：
root@N1:/tmp/upload# uname -a
Linux N1 5.4.145-flippy-64+o #319 SMP PREEMPT Sun Sep 12 21:31:11 CST 2021 aarch64 GNU/Linux
root@N1:/tmp/upload# opkg install luci-app-v2ray_2.0.0-1_all.ipk
Unknown package 'luci-app-v2ray'.
Collected errors:

pkg_hash_check_unresolved: cannot find dependency iptables-mod-tproxy for luci-app-v2ray
pkg_hash_fetch_best_installation_candidate: Packages for luci-app-v2ray found, but incompatible with the architectures configured
opkg_install_cmd: Cannot install package luci-app-v2ray.
请问哪里能找到这个合适版本的iptables-mod-tproxy，谢谢

我找到这个：iptables-mod-tproxy_1.8.3-1_aarch64_generic.ipk，还是说架构不对 :-(

ophub, 2021-09-16:
This needs to be included at compile time. If you need it, you can `make menuconfig` locally and confirm the `.config` file.

searchingx, 2021-09-17:
> This needs to be included at compile time. If you need it, you can `make menuconfig` locally and confirm the `.config` file.

OK, Thanks

closed 2021-09-17 by searchingx

szhhq, 2021-11-25:
章鱼星球上也遇到同样的问题。从系统软件仓库安装iptables-mod-tproxy失败。
上传ipk后用-force-depends强制安装能安装上，但自动下载安装的依赖的版本号很多是5.10的，可是系统内核是5.4的。
如果说是编译固件时就要包含进去，那以后想要安排别的软件可能还会有类似的问题呀。