ophub/amlogic-s9xxx-openwrt #127: S905w compile issue: [  Error  ] Have no this firmware: [ s905w - 5.12.13 ]

LeosSire, 2021-09-21:
I have successfully compiled for S905x2 in the past but I am now trying with S905w and I am getting the below message. Where do i get the the S905w specific kernel from? 

```
sudo ./make -d -b s905w -k 5.12.13
Welcome to use the OpenWrt packaging tool! 

Server space usage before starting to compile: 
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sda5      ext4  457G  221G  213G  51% / 

 firmware   ==>   openwrt-armvirt-64-default-rootfs.tar.gz
 rootsize   ==>   1024
 make_path   ==>   /home/dev/amlogic-s9xxx-openwrt

(1.1) Start packaging OpenWrt [ s905w - 5.12.13 ]. Remaining space is 214G.
 [  s905w  -  5.12.13  ]  (1/6) extract armvirt files.
 [  s905w  -  5.12.13  ]  (2/6) extract armbian files.
 [  s905w  -  5.12.13  ]  (3/6) refactor related files.
 [  Error  ] Have no this firmware: [ s905w - 5.12.13 ]
```

LeosSire, 2021-09-21:
Fixed it using the most recent code release.

ophub, 2021-09-21:
The version of the [kernel](https://github.com/ophub/flippy-kernel/tree/main/library) library must be specified

closed 2021-09-22 by ophub