ophub/amlogic-s9xxx-openwrt #129: enable wifi

kokenot, 2021-09-25:
how can I enable wireless feature . there is no wireless menu inside luci control panel

ophub, 2021-09-25:
What equipment are you? Unlimited use is enabled by default, but some devices do not have wireless drivers, such as x96max+

kokenot, 2021-09-25:
> What equipment are you? Unlimited use is enabled by default, but some devices do not have wireless drivers, such as x96max+

I can see wireless menu when using compiling script from openwrt_packit. I do know some device don't have driver for built in wifi. but we can use usb wifi as alternative. the problem is wireless menu is missing from luci panel. is it hide or any trick to enable it back?

ophub, 2021-09-26:
If there is a corresponding external wireless device driver, it will be automatically listed and displayed in the wireless management module.

kokenot, 2021-09-26:
what i mean is,there is no wireless menu in luci, even if i plug in the usb device. usb wifi driver is detected. also the wireless menu can't show. missing in luci.

ophub, 2021-09-27:
In my device, wireless is automatically displayed. I don't know the reason for your question. Sorry for not being able to help you.

closed 2021-09-27 by ophub