ophub/amlogic-s9xxx-openwrt #130: Installing to Odroid N2+ (Amlogic S922X)

locussss, 2021-09-30:
Hi,

I am trying to install your OpenWRT image on a Odroid N2+ with an Amlogic S922X chip.
I have tried all releases without luck, it boots with a lot of kernel messages and around the 80 seconds mark it sits and sits.
On the Odroid the blue led still gives a heartbeat.

Is there anything I am doing wrong? I have adjusted the file uEnv.txt with:
FDT=/dtb/amlogic/meson-g12b-odroid-n2-plus.dtb

Could you give me any tips?
Thanks!

ophub, 2021-09-30:
Don't modify uEnv.txt, start with the default s922x.

https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_lede_2021.10.01.0108/openwrt_s922x_k5.4.150_2021.10.01.0105.img.gz

locussss, 2021-10-01:
Will do!
I' report back....

locussss, 2021-10-01:
Hi!

I wrote the img from your link onto my microSDXC card and booted the Odroid N2+.
It got to the 48 seconds mark and then sits and sits.
The only difference, the blue led was on without heartbeat, so steady on.

I will attach a screenshot where it halted

[image]

anbachperu, 2021-10-01:
Hi I just tried to install on my x96 mini but I get this error help
[image]

anbachperu, 2021-10-01:
> Hi!
> 
> I wrote the img from your link onto my microSDXC card and booted the Odroid N2+. It got to the 48 seconds mark and then sits and sits. The only difference, the blue led was on without heartbeat, so steady on.
> 
> I will attach a screenshot where it halted
> 
> [image]

Did you manage to solve it?

ophub, 2021-10-02:
From the picture, there is no error, you have successfully started, just visit from the browser.

ophub, 2021-10-02:
You connect the keyboard to the box, press Enter, and the banner should be displayed.

locussss, 2021-10-02:
Hi!

Okay, I am in!
Through the webinterface I can access the Odroid. Yeah.
Thanks for that.

One last question:  I have an TP-link UE300 USB ethernet adapter connected to one of the USB 3 ports but it doen't show up with:
ip link show
How do I get the UE300 USB ethernet adapter up and running?

ophub, 2021-10-02:
Support rtl8152, rtl8153, rtl8156

locussss, 2021-10-02:
I understand it is supported but it isn't automatically up when I connect the adapter.
It should I think or am I wrong?
What command is necessary to get the adapter up and visible to make it work?
Thanks!

ophub, 2021-10-02:
Supported devices will be activated automatically, and you can restart it.

locussss, 2021-10-03:
Well, it was a matter of enabling all ethernet devices through petiteboot.
After that and booting up the UE3 ethernet adapter was there and active.
So, it is resolved and I will close the issue.
Thanks again!

closed 2021-10-03 by locussss