ophub/amlogic-s9xxx-openwrt #131: Installation on X96mini (Amlogic S905w)

anbachperu, 2021-10-01:
Hello I am trying to install it on my x96-mini Amlogic S905w but I have no success, when loading the white letters it reaches a point that it stops and does not continue, I explore the dbt / amlogic folder and I cannot find the .dbt file for the x96 mini I only find for the tx3 mini, if you could help me in that it would be very kind thanks
[image]
[image]

ophub, 2021-10-02:
From the picture, you have already started, just log in with your browser.

`meson-gxl-s905w-p281.dtb` can also be used.

The s905w box can only use the `5.4.y` kernel, for example: 
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_lede_2021.10.01.0108/openwrt_s905w_k5.4.150_2021.10.01.0101.img.gz

anbachperu, 2021-10-02:
Thank you very much, I was able to start, but it does not give me a Wi-Fi signal only through ethernet cable. Any configuration that I should make?

> From the picture, you have already started, just log in with your browser. You connect the keyboard to the box, press Enter, and the banner should be displayed.
> 
> `meson-gxl-s905w-p281.dtb` can also be used.
> 
> The s905w box can only use the `5.4.y` kernel, for example: https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_lede_2021.10.01.0108/openwrt_s905w_k5.4.150_2021.10.01.0101.img.gz

Thank you very much, I was able to start, but it does not give me a Wi-Fi signal only through ethernet cable. Any configuration that I should make?

ophub, 2021-10-02:
The wireless wifi of this box is not driven, only wired.

anbachperu, 2021-10-02:
> El wifi inalámbrico de esta caja no es accionado, solo cableado.

ok, how can i activate usb reception for simcard?

ophub, 2021-10-02:
External USB Wifi, you can choose from several products from ralink and medietek, E.g:

RT2770, RT2870 & RT3070, RT3071 & RT3072,RT3370, RT3572,RT3573 ,RT5370, RT5572，mt7601u, mt7610u, mt7612u

`rt` is basically an 802.11n device, and `USB2.0` is available; 

`mt7610u` and mt7612u can support 802.11ac, and `USB3.0` is required.

！！！Realtek does not support at all.

ophub, 2021-10-02:
> ok, how can i activate usb reception for simcard?

Select `dhcp` mode when sim card internet device connects to `wan` port.

anbachperu, 2021-10-02:
> > vale, ¿cómo puedo activar la recepción USB para simcard?
> 
> Select `dhcp` mode when sim card internet device connects to `wan` port.

I got here
[image]

anbachperu, 2021-10-02:
> > ok, how can i activate usb reception for simcard?
> 
> Select `dhcp` mode when sim card internet device connects to `wan` port.

can you install v2ray atau libernet atau dempul?

ophub, 2021-10-02:
The increase or decrease of the personalized software package can be compiled by yourself according to the instructions.

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md

anbachperu, 2021-10-02:
> The increase or decrease of the personalized software package can be compiled by yourself according to the instructions.
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md

And future updates can habilitate the wifi in this box?

ophub, 2021-10-02:
Openwrt does not support the wireless module of this box, and there is no driver.

anbachperu, 2021-10-02:
> Openwrt does not support the wireless module of this box, and there is no driver.

Do you know how to operate the s905x non-metallic module?

ophub, 2021-10-02:
I don’t know

closed 2021-10-03 by ophub